jkjung-avt/tensorrt_demos #260: MTCNN for higher resolutions

Fizmath, 2020-10-13:
Hello

I built  and  ran your MTCNN ``trt_mtcnn.py`` on my GTX 1050  laptop :  40 faces were detected out of 42 , with `` 12 fps``  .  The input image was of resolution 1920*1280 .  Later on i suspected the performance and realized that the ``*.jpg``  image was resized internally by an upper bound  ``1280*720``  :

https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/mtcnn.py#L470 

Is the release of this upper bound  straight forward ?  what changes should we make?

jkjung-avt, 2020-10-13:
> the *.jpg image was resized internally by an upper bound 1280*720 : https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/mtcnn.py#L470

That's right.  I needed to set a fixed (max) size for PNet in order to build a TensorRT engine for it (TensorRT 6.x- did not support dynamic input shapes).  So I just picked 1280x720.

In order to modify that, you'll have to read and understand [Optimizing TensorRT MTCNN](https://jkjung-avt.github.io/optimize-mtcnn/).  Then redesign the "stacking" of different scales.  You could check my example calculations here.

https://github.com/jkjung-avt/tensorrt_demos/blob/master/mtcnn/det1_relu.prototxt#L8-L24

Fizmath, 2020-10-13:
Thank you for your instant response.

My aim is by making use of :

https://docs.nvidia.com/deeplearning/triton-inference-server/master-user-guide/docs/model_configuration.html#delayed-batching

achieve this architecture  : 

Batch face detection from multiple 1080 cameras  
Batch  face  detection from multiple  1920 cameras 
Batch recognition  of the detected  faces 

and they all share the same GPU !  

[image]

Would you please give me more advice ?

jkjung-avt, 2020-10-14:
I don't have any experience on the Triton Inference Server, so I can't help you on that part.

Regardless, for your original question about modifying PNet inference code to detect faces in a larger resolution image, you still need to redesign the image patches and PNet input dimension.  It makes sense to get that working for single batch on your local machine first, before trying to port it over to Triton Inference Server.

closed 2020-10-21 by jkjung-avt