jkjung-avt/tensorrt_demos #261: YOLOv4 TensorRT model detection results are None

HtutLynn, 2020-10-16:
Hi, I've been working with your old codebase for YOLOv4 TensorRT model conversion and stuff. I just pulled the codebase for the latest commits and tried to convert YOLOv4 tiny 288 to TensorRT with exact steps described and do inference with it but the model is not detecting anything. After that, I checked outputs of the TensorRT runtime model and the inferred NumPy array turns out to be just zeros. 

```
[array([0., 0., 0., ..., 0., 0., 0.], dtype=float32), array([0., 0., 0., ..., 0., 0., 0.], dtype=float32), array([0., 0., 0., ..., 0., 0., 0.], dtype=float32)]
```
Is there any fix to it? I've been trying to deploy the YOLOV4-tiny-288 TensorRT model on Jetson Nano 2GB.

jkjung-avt, 2020-10-16:
1. Do you see the same problem if you start with a clean copy of the latest tensorrt_demos code?

2. Have you tested other models, say, yolov3-tiny-288 or yolov4-288?  Are you seeing the same problem too?

HtutLynn, 2020-10-19:
Hi, sorry for the late reply.
1. To be extra sure, I cloned a new repository in a different directory and followed the steps as described in the YOLOv4 and YOLOv3 section but it is still the same. No detected results or whatsoever. 
2. I tried for all of YOLOv3 and YOLOv4 variants but the result is still the same.

jkjung-avt, 2020-10-19:
1. What platform/GPU are you using?  Jetson or x86_64?
2. Which version of TensorRT?

HtutLynn, 2020-10-20:
Hi,
1. I am running on Nvidia GTX 1070. But I haven't tested it yet on Jetson Nano so I will try it soon. I tried on another machine with Nvidia GTX 2070 but it works on that machine so I am not sure about what is the error here.
2. TensorRT 7.1 (GTX 2070 machine has the same TensorRT version.)

jkjung-avt, 2020-10-20:
> Nvidia GTX 1070

Try update NVIDIA driver on this machine.

jkjung-avt, 2020-10-25:
Has this issue been resolved?

HtutLynn, 2020-10-26:
@jkjung-avt , I am sorry for the late reply. I haven't updated the Nvidia driver on my machine because I am kinda afraid of breaking things so I tested on Jetson Nano 4GB and it works fine as expected. I get around the same performance as you reported in YOLOv4 performance table section when I test with yolov4-288.trt . However, when I try to do inference with yolov4-tiny-288.trt on Nano, the performance is only around  ~9FPS compared to the performance on the table, 36.6FPS. That's a pretty big difference. Jetson Nano is flashed with Jetpack 4.4 and trt_yolo.py is run on a container from NGC. I followed every step that's mentioned on README so I am not sure what is causing this performance difference.

jkjung-avt, 2020-10-26:
If you set up your Jetson Nano the same way as how I did it, you should get similar results as mine.  I documented how I set up my Jetson Nano DevKit with JetPack-4.4 in the [JetPack-4.4 for Jetson Nano](https://jkjung-avt.github.io/jetpack-4.4/) blog post.  In particular, the versions/configurations of OpenCV and "protobuf" might affect the result.  And there could be certainly other stuffs which cause differences.

s5248, 2020-10-28:
Hi,
      The same problem happened in my setting, "Tesla K40m, TensorRT-7.0.0.11,        Driver Version: 418.39       CUDA Version: 10.1". Thanks for your reply

jkjung-avt, 2020-10-28:
> The same problem happened in my setting, "Tesla K40m, TensorRT-7.0.0.11, Driver Version: 418.39 CUDA Version: 10.1". Thanks for your reply

Referring to [TensorRT Release 7.x.x](https://docs.nvidia.com/deeplearning/tensorrt/release-notes/tensorrt-7.html#tensorrt-7), "Kepler" (CUDA Compute 3.x) has been deprecated in TensorRT 7.x.  Please test with a more modern GPU.

s5248, 2020-10-28:
> Referring to [TensorRT Release 7.x.x](https://docs.nvidia.com/deeplearning/tensorrt/release-notes/tensorrt-7.html#tensorrt-7), "Kepler" (CUDA Compute 3.x) has been deprecated in TensorRT 7.x. Please [...]

Sorry，I have no more modern GPU, maybe I should try TensorRT 6.x or TensorRT 5.x.
By the way, I checked the caffe samples in "TensorRT-7.0.0.11/samples/python/introductory_parser_samples", the results seems to be ok.

jkjung-avt, 2020-10-28:
> By the way, I checked the caffe samples in "TensorRT-7.0.0.11/samples/python/introductory_parser_samples", the results seems to be ok.

I think yolov4 model uses some layers which are not present in the "introductory_parser_samples" (resnet50).  If you don't have a newer GPU for testing, I think you should verify the result with TensorRT 6 instead.

hellosummer8850, 2020-10-30:
@jkjung-avt ,  Anyone solved? I felt this project is the most friendly tutorials of trt use, but I found that I have tested on Ubuntu18.04 x86-64 gtx1060, trt 7.1.3.4 + cuda11.0 + cudnn8. almost the latest GPU environments, but it still no result shown QAQ.

jkjung-avt, 2020-10-30:
I would still suggest to update your NVIDIA driver.  I use the following versions on my RTX-2080 Ti and GTX-1080 and it works.

* NVIDIA driver: 450.80.02
* CUDA: 10.2
* cuDNN: 8.0.1.13
* TensorRT: 7.1.3.4

hellosummer8850, 2020-11-01:
@jkjung-avt Amazing! it works, thanks a lot~

closed 2020-11-01 by jkjung-avt