jkjung-avt/tensorrt_demos #262: "Failed to open camera!"

kylelscott, 2020-10-16:
Hello. I am having an error when I run a yolov4-tiny-416. when I run this line 
`python3 trt_yolo.py --image ~/Pictures/dogs.jpg -m yolov4-tiny-416` to test the model on a picture saved as dogs.jpg I get this error
`ERROR: failed to open camera!`
I'm not using a live feed from a camera though, just a picture. Do you know why this error is happening?

jkjung-avt, 2020-10-17:
Please use the following command instead.  ("cv2" does not seem to handle "~" in the path properly.)

```
python3 trt_yolo.py --image ${HOME}/Pictures/dogs.jpg -m yolov4-tiny-416
```

kylelscott, 2020-10-18:
Unfortunately, I got the same error as before when I ran the command as follows:

`python3 trt_yolo.py --image ${HOME}/Picures/dogs.jpg -m yolov4-tiny-416`

jkjung-avt, 2020-10-18:
I suggest you to copy the jpg file to **current working directory**, to avoid path related issues.  Also make sure the file name is correct.  (Should it be "dog.jpg"?)

```
python3 trt_yolo.py --image dogs.jpg -m yolov4-tiny-416
```

jkjung-avt, 2020-10-25:
Has this issue been resolved?

kylelscott, 2020-10-25:
Hello. I tried it last night and it didn't work. I also tried deleting the
workspace and creating a new one with a newly cloned GitHub repo and that
one had build errors as it was trying to pull some colcon environmental
variables from the previous ws that I had just deleted. I tried changing
the variable paths but didn’t know how to access the file it was declaring
the paths in.

jkjung-avt, 2020-10-26:
Would you be able to test with a video file (--video) instead?

Or USB webcam (--usb)?

kylelscott, 2020-10-27:
Hello, sorry for the late reply. I was not able to test with the video or
usb args though.

jkjung-avt, 2020-10-27:
I think you need to verify your "cv2" is working first.  For example, make sure you have the "dog.jpg" file in the current directory and try executing the following code.  You should see the image displayed in a window.  If not, you need to fix your python3 "cv2" installation before continuing testing the code in this repo.

```
import cv2

img = cv2.imread('dog.jpg')
cv2.imshow('image', img)
cv2.waitKey(0)
```

kylelscott, 2020-10-29:
Hello, 
Thank you for your reply. I verified that opencv was working, and when I tried the command again it worked. I could also run it on the picture I mentioned earlier and with a USB camera. Thank you for your help, and this useful repo.

closed 2020-10-29 by kylelscott