jkjung-avt/tensorrt_demos #263: run as service

PatrikPatrik2, 2020-10-17:
Sorry for bringing this up again, but I have am not able to solve the problem with the script crashing after a day or so, stating "img returned null" (or similar). As a last chance for solving this I am will try to run the script in systemd, and get it to autostart when it crashes. However, it seems that I can not start a script with arguments in systemd. I have tried to hardcode my args into your code, but I only get errors. This is the args that I use: 
 --model ssd_mobilenet_v1_coco --rtsp --uri rtsp://username:password@192.168.2.46:554/h265/ch1/main/av_stream --height 1088 --width 1920

I would be very gratful if you can guide me how to hard code this into your code

Thanks

jkjung-avt, 2020-10-17:
1. It should be simple to solve the systemd issue.  You could put the command to run the trt_ssd application, with all necessary arguments, in a shell script.  Then just execute the script, which does not require any arguments, from your systemd service.

2. Otherwise, you could replace argparse.ArgumentParser (args) with something like a SimpleNameSpace or a class, and put your hardcoded arguments in there.  For example,

   ```
   from types import SimpleNamespace
   from utils.camera import Camera

   args = SimpleNamespace()
   args.image = 'dog.jpg'
   args.width = 1280
   args.height = 720
   args.copy_frame = False
   args.do_resize = False
   args.video_looping = False
   cam = Camera(args)
   ......
   ```

PatrikPatrik2, 2020-10-18:
Thanks, now I can start the script without the args, but I get an error when I use it in systemd

`pi@jetson:~/.config/systemd/user$ systemctl --user status python_demo_service
● python_demo_service.service - Python Demo Service
   Loaded: loaded (/home/pi/.config/systemd/user/python_demo_service.service; static; vendor preset: enabled)
   Active: active (running) since Sun 2020-10-18 15:53:18 CEST; 7s ago
 Main PID: 10808 (python3)
   CGroup: /user.slice/user-1000.slice/user@1000.service/python_demo_service.service
           ├─10808 /usr/bin/python3 /home/pi/SSD/PersonDetectionService/trt_ssd_10.6.py
           ├─10833 sh -c /usr/lib/aarch64-linux-gnu/libproxy/0.4.14/pxgsettings org.gnome.system.proxy org.gnome.system.proxy.http org.gnome.system.proxy.https org.gnome.system.proxy.ftp org.gnome.system.
           └─10835 /usr/lib/aarch64-linux-gnu/libproxy/0.4.14/pxgsettings org.gnome.system.proxy org.gnome.system.proxy.http org.gnome.system.proxy.https org.gnome.system.proxy.ftp org.gnome.system.proxy.

okt 18 15:53:25 jetson python3[10808]: Traceback (most recent call last):
okt 18 15:53:25 jetson python3[10808]:   File "/home/pi/SSD/PersonDetectionService/trt_ssd_10.6.py", line 378, in <module>
okt 18 15:53:25 jetson python3[10808]:     main()
okt 18 15:53:25 jetson python3[10808]:   File "/home/pi/SSD/PersonDetectionService/trt_ssd_10.6.py", line 364, in main
okt 18 15:53:25 jetson python3[10808]:     trt_ssd = TrtSSD(args.model, INPUT_HW)
okt 18 15:53:25 jetson python3[10808]:   File "/home/pi/SSD/PersonDetectionService/utils/ssd.py", line 84, in __init__
okt 18 15:53:25 jetson python3[10808]:     self.engine = self._load_engine()
okt 18 15:53:25 jetson python3[10808]:   File "/home/pi/SSD/PersonDetectionService/utils/ssd.py", line 56, in _load_engine
okt 18 15:53:25 jetson python3[10808]:     with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:
okt 18 15:53:25 jetson python3[10808]: FileNotFoundError: [Errno 2] No such file or directory: 'ssd/TRT_ssd_mobilenet_v1_coco.bin'

pi@jetson:~/.config/systemd/user$ systemctl --user status python_demo_service
● python_demo_service.service - Python Demo Service
   Loaded: loaded (/home/pi/.config/systemd/user/python_demo_service.service; static; vendor preset: enabled)
   Active: failed (Result: core-dump) since Sun 2020-10-18 15:53:26 CEST; 1s ago
  Process: 10808 ExecStart=/usr/bin/python3 /home/pi/SSD/PersonDetectionService/trt_ssd_10.6.py (code=dumped, signal=ABRT)
 Main PID: 10808 (code=dumped, signal=ABRT)

okt 18 15:53:26 jetson python3[10808]: PyCUDA ERROR: The context stack was not empty upon module cleanup.
okt 18 15:53:26 jetson python3[10808]: -------------------------------------------------------------------
okt 18 15:53:26 jetson python3[10808]: A context was still active when the context stack was being
okt 18 15:53:26 jetson python3[10808]: cleaned up. At this point in our execution, CUDA may already
okt 18 15:53:26 jetson python3[10808]: have been deinitialized, so there is no way we can finish
okt 18 15:53:26 jetson python3[10808]: cleanly. The program will be aborted now.
okt 18 15:53:26 jetson python3[10808]: Use Context.pop() to avoid this problem.
okt 18 15:53:26 jetson python3[10808]: -------------------------------------------------------------------
okt 18 15:53:26 jetson systemd[7191]: python_demo_service.service: Main process exited, code=dumped, status=6/ABRT
okt 18 15:53:26 jetson systemd[7191]: python_demo_service.service: Failed with result 'core-dump'.
pi@jetson:~/.config/systemd/user$ `

For the service I only use:
[Unit]
Description=Python Demo Service

[Service]
ExecStart=/usr/bin/python3 /home/pi/SSD/PersonDetectionService/trt_ssd_10.6.py

Can you understand why I can start trt_ssd_10.6.py in the terminal, but I get errors when I start is as a service?

jkjung-avt, 2020-10-18:
> okt 18 15:53:25 jetson python3[10808]: FileNotFoundError: [Errno 2] No such file or directory: 'ssd/TRT_ssd_mobilenet_v1_coco.bin'

This means the program could not find the TensorRT engine file.

You could consider putting the file at a fixed location and use an absolute path in the code:

https://github.com/jkjung-avt/tensorrt_demos/blob/e039e0824876d46443aa19e9ad7cf8f7723c713e/utils/ssd.py#L54

PatrikPatrik2, 2020-10-19:
Now it works! Thank you very much!!!!!!!!!!

closed 2020-10-19 by PatrikPatrik2