jkjung-avt/tensorrt_demos #264: How to get this map?

hjlee9182, 2020-10-19:
[image]

[image]

In yolov4 paper, yolov4 map is almost 44.

But you got over 70. How to get that point?
And What is difference between this result and paper?

jkjung-avt, 2020-10-19:
Thanks for asking.

According to the original [YOLOv4 paper](https://arxiv.org/pdf/2004.10934.pdf), AP and AP_50 of "yolov4-608" are **43.5%** and **65.7%** respectively.  You could refer to my [Verifying mAP of TensorRT Optimized SSD and YOLOv3 Models](https://jkjung-avt.github.io/trt-detection-map/) post for a better understanding about AP (mAP @ IoU=0.5:0.95) and AP_50 (mAP @ IoU=0.5).

[image]

In contrast, when I used my own [eval_yolo.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/eval_yolo.py) to verify mAP of my TensorRT yolov4-608 (FP16) engine, I got **0.488** and **0.736**.  Those numbers were slightly higher than what's in the original paper.

I myself am not 100% sure why.  But I think it might be related to how non-maximum suppression (NMS) should have been done when I evaluate mAP of the engines.

hjlee9182, 2020-10-19:
ok thanks!! :)

closed 2020-10-19 by hjlee9182