jkjung-avt/tensorrt_demos #265: CUDA error when using Camera with traitlets

xpeedster, 2020-10-19:
Hi,

I tried using as the input source for the trt_yolo model a Camera class with Python traitlets as the one used on the [Jetbot ](https://github.com/waveshare/jetbot/blob/master/jetbot/camera.py)project.

In order to do this the "loop_and_detect" function has been changed to be a "traitlets" callback (which means making the arguments that "loop_and_detect" takes Global variables, so that they dont need to be passed as arguments, because callback functions can't take arguments).
The image taken is valid and, if no detection is done(this means, commenting the " boxes, confs, clss = trt_yolo.detect(img, conf_th" line, it works. However, when uncommenting it, we get the error "cuda-error-in-nchwtonchhw2-33-invalid-resource-handle". Do you know why this can be?

Thanks for the help.

jkjung-avt, 2020-10-19:
I think this issue is similar to https://github.com/jkjung-avt/tensorrt_demos/issues/213.  The callback is executed in a child thread.  You need to specifically tell TrtYOLO to use the default CUDA context in this case (add "cuda_ctx=pycuda.autoinit.context" as shown below).

```
    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num, cuda_ctx=pycuda.autoinit.context)
```

xpeedster, 2020-10-19:
Thanks for your help. Sorry for the late response. It worked!

One thing, why did it work? I mean, what does the context do?

Thanks a lot!

jkjung-avt, 2020-10-19:
Please read CUDA and PyCUDA documentation.

xpeedster, 2020-10-19:
Oh, yes, sorry.

Thanks for everything, I close the issue.

closed 2020-10-19 by xpeedster