jkjung-avt/tensorrt_demos #266: pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument

Darshcg, 2020-10-19:
Hi, @jkjung-avt,

I am using trt_yolo.py to run inference on my Custom Tensorrt model. I did the changes for my Model, but I am facing an error as **pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument**. Can you please assist me to get rid of this? What exactly is the reason behind this?

Complete Log:
```
File "/home/smartcow/SL-PMH/TRTFPMH.py", line 206, in detect
    stream=self.stream)
  File "/home/smartcow/SL-PMH/TRTFPMH.py", line 115, in do_inference_v2
    [cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
  File "/home/smartcow/SL-PMH/TRTFPMH.py", line 115, in <listcomp>
    [cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument
```

Environment:
**TRT version: 7.1.3
CUDA Version: 10.2
Platform: Jetson Nano**

jkjung-avt, 2020-10-19:
> pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument

This usually means the sizes of the "host buffer" and the "device buffer" mismatch.

You should check whether you have resized your input images correctly: https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/yolo_with_plugins.py#L35

Darshcg, 2020-10-19:
Hi @jkjung-avt,

Following your response, I figured out that there was a mistake in Resize in the input image. Thanks a lot @jkjung-avt. With your materials, I had learned a lot. Thank you a lot for your support.

Btw, another thing is that my TRT model is 304.5 MB and the Process is being killed due to low memory in Jetson Nano while running an Inference call(https://github.com/jkjung-avt/tensorrt_demos/blob/e039e0824876d46443aa19e9ad7cf8f7723c713e/utils/yolo_with_plugins.py#L205).
I think the model size is heavy, maybe that's the reason the process is being killed. 

1. Upto what Model Size, the Jetson Nano can support for running an inference?
2. And how to reduce the size of the TRT model size(is there any way to reduce while converting from ONNX to TRT)?

Thanks

jkjung-avt, 2020-10-19:
> 1. Upto what Model Size, the Jetson Nano can support for running an inference?

I don't know.  You have to test to find it out, I guess.

> 2. And how to reduce the size of the TRT model size(is there any way to reduce while converting from ONNX to TRT)?

There's a technique called "model pruning", with which you could try to reduce size of the model before applying TensorRT optimization on it.  For example, NVIDIA's [Transfer Learning Toolkit](https://developer.nvidia.com/transfer-learning-toolkit) supports pruning of trained neural networks.

Darshcg, 2020-10-20:
Thank you a lot for your response. I will try to further optimize it further with Transfer Learning Toolkit.

closed 2020-10-20 by Darshcg