jkjung-avt/tensorrt_demos #268: TypeError: buffer is too small for requested array

sharoseali, 2020-10-23:
HI!   On converting after training custom data on YOLO_v4 model when I run the script yolo_to_onnx.py  `python3 yolo_to_onnx.py --model yolov4-608 --category_num 12`  I get the error pasted below. Please help how can fix this issue. I make some changes in filters numbers in cfg file for detecting small objects before training but I don't think so that here it contains some problem, as I am giving my cfg file to fully interpret my model structure. thanks

```
Parsing DarkNet cfg file...
Building ONNX graph...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 957, in <module>
    main()
  File "yolo_to_onnx.py", line 940, in main
    verbose=True)
  File "yolo_to_onnx.py", line 451, in build_onnx_graph
    params)
  File "yolo_to_onnx.py", line 323, in load_conv_weights
    conv_params, 'conv', 'weights')
  File "yolo_to_onnx.py", line 351, in _create_param_tensors
    conv_params, param_category, suffix)
  File "yolo_to_onnx.py", line 382, in _load_one_param_type
    buffer=self.weights_file.read(param_size * 4))
TypeError: buffer is too small for requested array
```

jkjung-avt, 2020-10-24:
Be sure to my [TensorRT YOLOv3 For Custom Trained Models](https://jkjung-avt.github.io/trt-yolov3-custom/) for how to adapt the tensorrt_demos code to your own custom trained yolov3/yolov4 models.

Otherwise, could you provide the .cfg and .weights files for analysis?

sharoseali, 2020-10-26:
> Be sure to my [TensorRT YOLOv3 For Custom Trained Models](https://jkjung-avt.github.io/trt-yolov3-custom/) for how to adapt the tensorrt_demos code to your own custom trained yolov3/yolov4 models.
> 
> Otherwise, could you provide the .cfg and .weights files for analysis?

Thanks, for replying @jkjung-avt  Here is my cfg file. 
```

[net]
batch=16
subdivisions=64
# Training
#width=512
#height=512
width=1120
height=928
channels=3
momentum=0.949
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.0013
burn_in=1000
max_batches = 30000
policy=steps
steps=400000,450000
scales=.1,.1

#cutmix=1
mosaic=1

#:104x104 54:52x52 85:26x26 104:13x13 for 416

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=mish

# Downsample

[convolutional]
batch_normalize=1
filters=64
size=3
stride=2
pad=1
activation=mish

[convolutional]
batch_normalize=1
filters=64
size=1
stride=1
pad=1
activation=mish

[route]
layers = -2

[convolutional]
batch_normalize=1
filters=64
size=1
stride=1
pad=1
[... 1058 lines cut ...]
[convolutional]
size=1
stride=1
pad=1
filters=51
activation=linear

[yolo]
mask = 6,7,8
anchors = 12, 16, 19, 36, 40, 28, 36, 75, 76, 55, 72, 146, 142, 110, 192, 243, 459, 401
classes=12
num=9
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1
scale_x_y = 1.05
iou_thresh=0.213
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
nms_kind=greedynms
beta_nms=0.6
max_delta=5
```

jkjung-avt, 2020-10-26:
I wonder whether the problem was caused by this "stride=4" upsample layer...  Could you also provide the .weights file?

```
[upsample]
stride=4
```

On the other hands, the following settings in your .cfg file might need to be fixed:

1. "subdivisions" (dividing each batch into this many subdivisions during training) should not be greater than "batch".

   ```
   [net]
   batch=16
   subdivisions=64
   ```

2. Values in "steps" should be less than "max_batches".  Otherwise, I think the "steps" are useless.

   ```
   max_batches = 30000
   policy=steps
   steps=400000,450000
   scales=.1,.1
   ```

sharoseali, 2020-10-26:
> I wonder whether the problem was caused by this "stride=4" upsample layer... Could you also provide the .weights file?
>
> ```
> [upsample]
> stride=4
> ```
>
> On the other hands, the following settings in [...]

@jkjung-avt. if I change stride then wıll there not be a problem again while parsing the weights because the model is already trained with thıs configuration file. Where can I share the weights . your email please. Thanks

jkjung-avt, 2020-10-26:
You could share the .weights using, say, Google Drive.  My email address is jkjung13@gmail.com

sharoseali, 2020-10-26:
> You could share the .weights using, say, Google Drive. My email address is [jkjung13@gmail.com](mailto:jkjung13@gmail.com)

I share it with you in your email. Please check

jkjung-avt, 2020-10-26:
I received the email.  I will test it out when I have a chance.

jkjung-avt, 2020-11-01:
I think the issue is indeed centered around the stride 4 upsample operation.

```
[upsample]
stride=4
```

It results in different shape for one of the output feature maps.  That is, the `[c, h //  8, w //  8]` shape should be modified to `[c, h //  4, w //  4]`.  Please modify the following 3 places in the source code.  Let me know if it solves the problem for you.

https://github.com/jkjung-avt/tensorrt_demos/blob/e039e0824876d46443aa19e9ad7cf8f7723c713e/yolo/yolo_to_onnx.py#L922-L924

https://github.com/jkjung-avt/tensorrt_demos/blob/e039e0824876d46443aa19e9ad7cf8f7723c713e/yolo/plugins.py#L46

https://github.com/jkjung-avt/tensorrt_demos/blob/e039e0824876d46443aa19e9ad7cf8f7723c713e/utils/yolo_with_plugins.py#L225

sharoseali, 2020-11-01:
> I think the issue is indeed centered around the stride 4 upsample operation.
>
> ```
> [upsample]
> stride=4
> ```
>
> It results in different shape for one of the output feature maps. That is, the `[c, h // 8, [...]

Thank You so much @jkjung-avt for replying. I will surely edit this.

jkjung-avt, 2020-11-09:
Closing this since there's no follow-up questions.

closed 2020-11-09 by jkjung-avt

sharoseali, 2020-11-10:
> Thank You so much @jkjung-avt for replying. I will surely edit this.

@jkjung-avt  sorry for the late reply. I make the changes, in yolo_to_onnx, and onnx_to_trt as suggested now, and both work fine
but when I test trt_yolo.py to test I get this error:
`Traceback (most recent call last):
  File "trt_yolo.py", line 139, in <module>
    main()
  File "trt_yolo.py", line 131, in main
    loop_and_detect(args.model, cam, trt_yolo, conf_th=0.3, vis=vis)
  File "trt_yolo.py", line 66, in loop_and_detect
    boxes, confs, clss = trt_yolo.detect(img, conf_th)
  File "/home/xavier/Desktop/tensorrt_demos_modified_models/utils/yolo.py", line 546, in detect
    for output, shape in zip(trt_outputs, self.output_shapes)]
  File "/home/xavier/Desktop/tensorrt_demos_modified_models/utils/yolo.py", line 546, in <listcomp>
    for output, shape in zip(trt_outputs, self.output_shapes)]
ValueError: cannot reshape array of size 551616 into shape (1,51,52,52)`

but I make a change in /utils/yolo.py" as you write before Now it worked fine: 
`def _init_yolov4_postprocessor(self):
        h, w = self.input_shape
        filters = (self.category_num + 5) * 3
        if 'tiny' in self.model:
            self.output_shapes = [(1, filters, h // 32, w // 32),
                                  (1, filters, h // 16, w // 16)]
        else:
            self.output_shapes = [(1, filters, h //  4, w //  4),
                                  (1, filters, h // 16, w // 16),
                                  (1, filters, h // 32, w // 32)]`
@jkjung-avt  I observe the some changes in detection, mAP of trt models after conversion is reducing from 2 to 10 %, why it is so, Is there any solution and 
After using these 3 scripts, Can you also tell in which state the new model is. I mean like FP32, FP16 or INT 8? 
Can we more optimize our model after trt format, as you mentioned something like DLA Cores in this repo?
Thanks for helping !

jkjung-avt, 2020-11-10:
> but I make a change in /utils/yolo.py" as you write before Now it worked fine:
> `def _init_yolov4_postprocessor(self): h, w = self.input_shape filters = (self.category_num + 5) * 3 if 'tiny' in [...]

You are running an older version of the code...  When you have a chance, I do recommend you to pull the latest code from this repo.  The new "yolo_layer" plugin implementation would produce higher frame rate (better inference speed).

> @jkjung-avt I observe the some changes in detection, mAP of trt models after conversion is reducing from 2 to 10 %, why it is so, Is there any solution

Refer to https://github.com/jkjung-avt/tensorrt_demos/issues/255.  I also observed reduced accuracy (mAP) on, say, the yolov3-tiny models.  I do not know why.

> After using these 3 scripts, Can you also tell in which state the new model is. I mean like FP32, FP16 or INT8?

FP16.

> Can we more optimize our model after trt format, as you mentioned something like DLA Cores in this repo?

Yes.  I already implemented INT8 and DLA core optimization in the latest code.  Please check out [Demo #6: Using INT8 and DLA core](https://github.com/jkjung-avt/tensorrt_demos#demo-6-using-int8-and-dla-core) in the README.

sharoseali, 2020-11-10:
> > but I make a change in /utils/yolo.py" as you write before Now it worked fine:
> > `def _init_yolov4_postprocessor(self): h, w = self.input_shape filters = (self.category_num + 5) * 3 if 'tiny' in self.model: self.output_shapes = [(1, filters, h // 32, w // 32), (1, filters, h // 16, w // 16)] else: self.output_shapes = [(1, filters, h // 4, w // 4), (1, filters, h // 16, w // 16), (1, filters, h // 32, w // 32)]`
> 
> You are running an older version of the code... When you have a chance, I do recommend you to pull the latest code from this repo. The new "yolo_layer" plugin implementation would produce a higher frame rate (better inference speed).
> 
> > @jkjung-avt I observe the some changes in detection, mAP of trt models after conversion is reducing from 2 to 10 %, why it is so, Is there any solution
> 
> Refer to #255. I also observed reduced accuracy (mAP) on, say, the yolov3-tiny models. I do not know why.
> 
> > After using these 3 scripts, Can you also tell in which state the new model is. I mean like FP32, FP16 or INT8?
> 
> FP16.
> 
> > Can we more optimize our model after trt format, as you mentioned something like DLA Cores in this repo?
> 
> Yes. I already implemented INT8 and DLA core optimization in the latest code. Please check out [Demo #6: Using INT8 and DLA core](https://github.com/jkjung-avt/tensorrt_demos#demo-6-using-int8-and-dla-core) in the README.

Thanks, @jkjung-avt  for guiding, I will surely pull the new version of this repo. I get an mAP reduction in yoloV4.
Also, I was reading the official documentation of Tensor RT where we can make and optımıze our model customly. I don't know how much in thıs repo we can tweak our network to get a hold of yolov4 or v3 completely or which ıs the best implımentation of Yolo models in TensorFlow because ıt seems to be a much deeper relation of TF wıth TRT towards optimization.

Shame-fight, 2020-11-14:
> HI! On converting after training custom data on YOLO_v4 model when I run the script yolo_to_onnx.py `python3 yolo_to_onnx.py --model yolov4-608 --category_num 12` I get the error pasted below. [...]

I have the same problem, and have modified 3 places [c, h // 8, w // 8] to [c, h // 4, w // 4], but the same appears when I run yolo_to_onnx.py error. It can be detected normally under the darknet framework, category 4.
At the same time, I have a single-category yolov3-416 model that has been converted to trt and can be used normally.
I will send the cfg and weight files to your mailbox, please help me to verify if there is a problem.
@jkjung-avt

sharoseali, 2020-11-17:
@jkjung-avt  in the Yolov3 cfg file I have 5 Yolo layers for better detection of all size objects (as suggested by Alexey), how can I edit number of yolo layers  here so that I cant get an error in yolo_with_plugins.py  `bad number of yolo layers 5 VS 3`
Thanks

jkjung-avt, 2020-11-17:
@sharoseali  You need to find the 5 output layers and the corresponding tensor shapes, then modify source code of these 3 places:

https://github.com/jkjung-avt/tensorrt_demos/blob/26fb5a441134b0a58bd5804a5c7cdd1183a14554/yolo/yolo_to_onnx.py#L914-L916

https://github.com/jkjung-avt/tensorrt_demos/blob/26fb5a441134b0a58bd5804a5c7cdd1183a14554/yolo/plugins.py#L41

https://github.com/jkjung-avt/tensorrt_demos/blob/26fb5a441134b0a58bd5804a5c7cdd1183a14554/utils/yolo_with_plugins.py#L221