jkjung-avt/tensorrt_demos #269: How to deploy YOLOV4 model to Triton Inference Server

AdamChrist, 2020-10-26:
Hi,I tried to deploy YOLOV4's TensorRT model to Triton Inference Server,but it did not succeed. Can you provide an example?

I have tried the method in [yolov4-triton-tensorrt](https://github.com/isarsoft/yolov4-triton-tensorrt) and it works, but the plugins has no way to support dynamic Class_NUM .I want to deploy two different models on Triton Inference Server, their Class_num is different.But in the plugin, Class_NUM is fixed.

I noticed that the plugin you wrote is optimized for parameters such as Class_NUM, but I have not successfully deployed.

Can you provide some help in this regard? Thank you very much!

jkjung-avt, 2020-10-26:
I have never studied or used Triton Inference Server, so I don't think I'd be able to help.

Otherwise, yes.  My "yolo_layer" plugin implementation supports the following which are not supported by NVIDIA's examples:

* arbitrary input width and height, e.g. 416x256.
* different numClasses for different models.
* arbitrary numbers (not limited to 3) of anchors for yolo_layer's.
* handling of "scale_x_y" (producing bbox coordinates which match the original darknet design).

AdamChrist, 2020-10-26:
@jkjung-avt Okay, your code has given me a lot of inspiration, thank you very much

closed 2020-10-26 by AdamChrist

alphapibeta, 2021-01-08:
> Hi,I tried to deploy YOLOV4's TensorRT model to Triton Inference Server,but it did not succeed. Can you provide an example?
>
> I have tried the method in [...]

<h1> Try this config </h1>
max_batch_size=16 in my case
<br>

config.pbtxt
```
name: "yolov4"
platform: "tensorrt_plan"
max_batch_size: 16
dynamic_batching {
  preferred_batch_size: [8,4]
  max_queue_delay_microseconds: 100
  
}
input [
  {
    name: "000_net"
    data_type: TYPE_FP32
    format: FORMAT_NCHW
    dims: [ 3, 608, 608 ]
  }
]
output [
  {
    name: "(Unnamed Layer* 506) [PluginV2IOExt]_output_0"
    data_type: TYPE_FP32
    dims: [121296,1,1]
    
  },
  {
    name: "(Unnamed Layer* 507) [PluginV2IOExt]_output_0"
    data_type: TYPE_FP32
    dims: [30324,1,1]
  },
  {
    name: "(Unnamed Layer* 508) [PluginV2IOExt]_output_0"
    data_type: TYPE_FP32
    dims: [7581,1,1]
  }
]
instance_group [
  {
    count:1
    kind: KIND_GPU
  }
]
```

IsraelLencina, 2021-02-23:
> > Hi,I tried to deploy YOLOV4's TensorRT model to Triton Inference Server,but it did not succeed. Can you provide an example?
> > I have tried the method in [yolov4-triton-tensorrt](https://github.com/isarsoft/yolov4-triton-tensorrt) and it works, but the plugins has no way to support dynamic Class_NUM .I want to deploy two different models on Triton Inference Server, their Class_num is different.But in the plugin, Class_NUM is fixed.
> > I noticed that the plugin you wrote is optimized for parameters such as Class_NUM, but I have not successfully deployed.
> > Can you provide some help in this regard? Thank you very much!
> 
> # Try this config
> max_batch_size=16 in my case
> 
> config.pbtxt
> 
> ```
> name: "yolov4"
> platform: "tensorrt_plan"
> max_batch_size: 16
> dynamic_batching {
>   preferred_batch_size: [8,4]
>   max_queue_delay_microseconds: 100
>   
> }
> input [
>   {
>     name: "000_net"
>     data_type: TYPE_FP32
>     format: FORMAT_NCHW
>     dims: [ 3, 608, 608 ]
>   }
> ]
> output [
>   {
>     name: "(Unnamed Layer* 506) [PluginV2IOExt]_output_0"
>     data_type: TYPE_FP32
>     dims: [121296,1,1]
>     
>   },
>   {
>     name: "(Unnamed Layer* 507) [PluginV2IOExt]_output_0"
>     data_type: TYPE_FP32
>     dims: [30324,1,1]
>   },
>   {
>     name: "(Unnamed Layer* 508) [PluginV2IOExt]_output_0"
>     data_type: TYPE_FP32
>     dims: [7581,1,1]
>   }
> ]
> instance_group [
>   {
>     count:1
>     kind: KIND_GPU
>   }
> ]
> ```

Can you send from a client only one request?