jkjung-avt/tensorrt_demos #271: ERROR: …/rtSafe/cuda/reformat.cu (925) - Cuda Error in NCHWToNCHHW2: 400 (invalid resource handle)

bhargavravat, 2020-10-27:
I have converted YOLOV3 model to onnx and from onnx to TRT model .

Now I am trying to run inference on received image using socketio . 
[trt_mask_plugin.txt](https://github.com/jkjung-avt/tensorrt_demos/files/5442737/trt_mask_plugin.txt)

> Blockquote[TensorRT] WARNING: Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.

> Blockquote[TensorRT] WARNING: Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.

> Blockquote [TensorRT] ERROR: FAILED_EXECUTION: std::exception

If I run on the local images , then model is working OKAY. However whenever I am trying to do inference on the image which is recieved from socketio connection at that time it is giving me above error. Can you please help me with this ? What modification do I need to do ?

bhargavravat, 2020-10-27:
Here is the complete details of board:

> NVIDIA Jetson Nano (Developer Kit Version)
> L4T 32.4.4 [ JetPack UNKNOWN ]
> Ubuntu 18.04.5 LTS
> Kernel Version: 4.9.140-tegra
> CUDA 10.2.89
> CUDA Architecture: 5.3
> OpenCV version: 4.1.1
> OpenCV Cuda: YES
> CUDNN: 8.0.0.180
> TensorRT: 7.1.3.0
> Vision Works: 1.6.0.501
> VPI: 0.4.4

jkjung-avt, 2020-10-27:
Refer to this: https://github.com/jkjung-avt/tensorrt_demos/issues/265#issuecomment-712006217.

I think the following might solve the problem for you:

```
trt_yolo = TrtYOLO("yolov3-custom-416", (416, 416), 3, cuda_ctx=pycuda.autoinit.context)
```

bhargavravat, 2020-10-27:
> Refer to this: [#265 (comment)](https://github.com/jkjung-avt/tensorrt_demos/issues/265#issuecomment-712006217).
> 
> I think the following might solve the problem for you:
> 
> ```
> trt_yolo = TrtYOLO("yolov3-custom-416", (416, 416), 3, cuda_ctx=pycuda.autoinit.context)
> ```

Awesome!! It worked for me!!! :+1:

closed 2020-10-27 by bhargavravat

bhargavravat, 2020-11-05:
@jkjung-avt  ; 

Hi can you help me with what changes do I need to make in order to do inference in multiple images at the same time ? 

As said earlier I am receiving images from socker io connection over the internet , I am able to do the inference on single image which is recieved . 

Now that I am getting 10 images at the same time and I want to process all of them at the same time what changes do I have to do in my codd ??

jkjung-avt, 2020-11-05:
https://github.com/jkjung-avt/tensorrt_demos/issues/249#issuecomment-705378621