jkjung-avt/tensorrt_demos #272: Field number 0 is illegal

jch-wang, 2020-10-27:
I trained my ssd_mobilenet_v2 model with code repository [https://github.com/tensorflow/models](url) and obtained the model file frozen_inference_graph.pb .  I just changed "INPUT_DIMS=(3,300,300)" in tensorrt_demos/ssd/build_engine.py to "INPUT_DIMS=(3,608,342)".  Then I use build_engine.py  to build the TRT engine .The following error occurred：
```
lqs@lqs-desktop:~/tensorrt_demos/ssd$ ./build_engines.sh
+ for model in ssd_mobilenet_v2_coco
+ python3 build_engine.py ssd_mobilenet_v2_coco
2020-10-26 23:13:08.055422: I tensorflow/stream_executor/platform/default/dso_loader.cc:48] Successfully opened dynamic library libcudart.so.10.2
WARNING:tensorflow:Deprecation warnings have been disabled. Set TF_ENABLE_DEPRECATION_WARNINGS=1 to re-enable them.
Traceback (most recent call last):
  File "build_engine.py", line 304, in <module>
    main()
  File "build_engine.py", line 279, in main
    gs.DynamicGraph(spec['input_pb']),
  File "/usr/lib/python3.6/dist-packages/graphsurgeon/StaticGraph.py", line 79, in __init__
    self.read(graphdef)
  File "/usr/lib/python3.6/dist-packages/graphsurgeon/StaticGraph.py", line 173, in read
    self._internal_graphdef.ParseFromString(frozen_pb.read())
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/message.py", line 199, in ParseFromString
    return self.MergeFromString(serialized)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/internal/python_message.py", line 1145, in MergeFromString
    if self._InternalParse(serialized, 0, length) != length:
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/internal/python_message.py", line 1212, in InternalParse
    pos = field_decoder(buffer, new_pos, end, self, field_dict)
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/internal/decoder.py", line 733, in DecodeRepeatedField
    if value.add()._InternalParse(buffer, pos, new_pos) != new_pos:
  File "/usr/local/lib/python3.6/dist-packages/google/protobuf/internal/python_message.py", line 1195, in InternalParse
    raise message_mod.DecodeError('Field number 0 is illegal.')
google.protobuf.message.DecodeError: Field number 0 is illegal.

```
Maybe you can give me some useful advice。

jkjung-avt, 2020-10-27:
Have you patched your "graphsurgeon" module?  Please refer to:

https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/install.sh#L16-L27

jch-wang, 2020-10-27:
> Have you patched your "graphsurgeon" module? Please refer to:
> 
> https://github.com/jkjung-avt/tensorrt_demos/blob/master/ssd/install.sh#L16-L27

I did this. And I do it again
```
lqs@lqs-desktop:~/tensorrt_demos/ssd$ ./install.sh
** Patch 'graphsurgeon.py' in TensorRT
[sudo] password for lqs: 
patching file /usr/lib/python3.6/dist-packages/graphsurgeon/node_manipulation.py
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored
** Making symbolic link of libflattenconcat.so
** Installation done
```
I build the engine again and get the same error. I don't have any clue about it.

jkjung-avt, 2020-10-27:
I have not encountered this error before...  Which version of tensorflow are you using?

jch-wang, 2020-10-30:
> I have not encountered this error before... Which version of tensorflow are you using?

It was probably because I trained in the window10 environment that there were errors in the model. Now in Ubuntu, I've got a model that doesn't report errors in the training process. My model detects 7 kinds of targets, and the network input size is 608 * 342. The build engine process did not produce file “TRT_ssd_mobilenet_v2_coco.bin”。

```
lqs@lqs-desktop:~/tensorrt_demos_yolov3_4/ssd$ ./build_engines.sh
+ for model in ssd_mobilenet_v2_coco
+ python3 build_engine.py ssd_mobilenet_v2_coco
2020-10-30 17:58:18.314257: I tensorflow/stream_executor/platform/default/dso_loader.cc:48] Successfully opened dynamic library libcudart.so.10.2
WARNING:tensorflow:Deprecation warnings have been disabled. Set TF_ENABLE_DEPRECATION_WARNINGS=1 to re-enable them.
NOTE: UFF has been tested with TensorFlow 1.15.0.
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
UFF Version 0.6.9
=== Automatically deduced input nodes ===
[name: "Input"
op: "Placeholder"
attr {
  key: "dtype"
  value {
    type: DT_FLOAT
  }
}
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: 1
      }
      dim {
        size: 3
      }
      dim {
        size: 608
      }
      dim {
        size: 342
      }
    }
  }
}
]
=========================================

Using output node NMS
Converting to UFF graph
Warning: No conversion function registered for layer: NMS_TRT yet.
Converting NMS as custom op: NMS_TRT
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:226: The name tf.AttrValue is deprecated. Please use tf.compat.v1.AttrValue instead.

Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_conf as custom op: FlattenConcat_TRT
Warning: No conversion function registered for layer: GridAnchor_TRT yet.
Converting MultipleGridAnchorGenerator as custom op: GridAnchor_TRT
Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_loc as custom op: FlattenConcat_TRT
DEBUG [/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:143] Marking ['NMS'] as outputs
No. nodes: 644
UFF Output written to /home/lqs/tensorrt_demos_yolov3_4/ssd/ssd_mobilenet_v2_coco.uff
UFF Text Output written to /home/lqs/tensorrt_demos_yolov3_4/ssd/ssd_mobilenet_v2_coco.pbtxt
[TensorRT] INFO: 
[TensorRT] INFO: --------------- Layers running on DLA: 
[TensorRT] INFO: 
[TensorRT] INFO: --------------- Layers running on GPU: 
[TensorRT] INFO: AnchorInput, MultipleGridAnchorGenerator, FeatureExtractor/MobilenetV2/Conv/Conv2D + FeatureExtractor/MobilenetV2/Conv/Relu6, FeatureExtractor/MobilenetV2/expanded_conv/depthwise/depthwise + FeatureExtractor/MobilenetV2/expanded_conv/depthwise/Relu6, FeatureExtractor/MobilenetV2/expanded_conv/project/Conv2D, FeatureExtractor/MobilenetV2/expanded_conv_1/expand/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_1/expand/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_1/depthwise/depthwise + FeatureExtractor/MobilenetV2/expanded_conv_1/depthwise/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_1/project/Conv2D, FeatureExtractor/MobilenetV2/expanded_conv_2/expand/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_2/expand/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_2/depthwise/depthwise + FeatureExtractor/MobilenetV2/expanded_conv_2/depthwise/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_2/project/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_2/add, FeatureExtractor/MobilenetV2/expanded_conv_3/expand/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_3/expand/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_3/depthwise/depthwise + FeatureExtractor/MobilenetV2/expanded_conv_3/depthwise/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_3/project/Conv2D, FeatureExtractor/MobilenetV2/expanded_conv_4/expand/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_4/expand/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_4/depthwise/depthwise + FeatureExtractor/MobilenetV2/expanded_conv_4/depthwise/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_4/project/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_4/add, FeatureExtractor/MobilenetV2/expanded_conv_5/expand/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_5/expand/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_5/depthwise/depthwise + FeatureExtractor/MobilenetV2/expanded_conv_5/depthwise/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_5/project/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_5/add, FeatureExtractor/MobilenetV2/expanded_conv_6/expand/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_6/expand/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_6/depthwise/depthwise + FeatureExtractor/MobilenetV2/expanded_conv_6/depthwise/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_6/project/Conv2D, FeatureExtractor/MobilenetV2/expanded_conv_7/expand/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_7/expand/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_7/depthwise/depthwise + FeatureExtractor/MobilenetV2/expanded_conv_7/depthwise/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_7/project/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_7/add, FeatureExtractor/MobilenetV2/expanded_conv_8/expand/Conv2D + FeatureExtractor/MobilenetV2/expanded_conv_8/expand/Relu6, FeatureExtractor/MobilenetV2/expanded_conv_8/depthwise/depthwise + FeatureExtractor/Mo [... cut ...] idAnchorGenerator_2 copy, MultipleGridAnchorGenerator_3 copy, MultipleGridAnchorGenerator_4 copy, MultipleGridAnchorGenerator_5 copy, NMS, 
[TensorRT] INFO: Some tactics do not have sufficient workspace memory to run. Increasing workspace size may increase performance, please check verbose output.
[TensorRT] INFO: Detected 1 inputs and 2 output network tensors.
#assertionnmsPlugin.cpp,246
./build_engines.sh: line 5: 18172 Aborted                 (core dumped) python3 build_engine.py ${model}
```

jkjung-avt, 2020-10-30:
> My model detects 7 kinds of targets, and the network input size is 608 * 342.

If your model is not using 300x300 inputs, you'll have to modify `featureMapShapes` accordingly.

https://github.com/jkjung-avt/tensorrt_demos/blob/e039e0824876d46443aa19e9ad7cf8f7723c713e/ssd/build_engine.py#L172

In fact, the "gridAnchorPlugin" provided by NVIDIA probably does not support non-square grid anchors by default.  In order to make it work, I think you'd need to:

1. build the [gridAnchorPlugin](https://github.com/NVIDIA/TensorRT/tree/master/plugin/gridAnchorPlugin) by youself,
2. use `GridAnchorRect_TRT` as the plugin name, and
3. use 2 numbers for each grid anchor layer (instead of 1).  Refer to source code below for details.

https://github.com/NVIDIA/TensorRT/blob/master/plugin/gridAnchorPlugin/gridAnchorPlugin.cpp#L464-L480

jch-wang, 2020-10-30:
> In fact, the "gridAnchorPlugin" provided by NVIDIA probably does not support non-square grid anchors by default. In order to make it work, I think you'd need to:
> 
> 1. build the [gridAnchorPlugin](https://github.com/NVIDIA/TensorRT/tree/master/plugin/gridAnchorPlugin) by youself,

For input size 608 * 342, I modified it：
`featureMapShapes=[(38, 22), (19, 11), (10, 6), (5, 3), (3, 2), (2, 1)]`
Then I build the engine file again and the following message appears：
```
lqs@lqs-desktop:~/tensorrt_demos_yolov3_4/ssd$ ./build_engines.sh
+ for model in ssd_mobilenet_v2_coco
+ python3 build_engine.py ssd_mobilenet_v2_coco
2020-10-30 22:10:59.075633: I tensorflow/stream_executor/platform/default/dso_loader.cc:48] Successfully opened dynamic library libcudart.so.10.2
WARNING:tensorflow:Deprecation warnings have been disabled. Set TF_ENABLE_DEPRECATION_WARNINGS=1 to re-enable them.
Traceback (most recent call last):
  File "build_engine.py", line 252, in <module>
    main()
  File "build_engine.py", line 229, in main
    spec)
  File "build_engine.py", line 121, in add_plugin
    numLayers=6
  File "/usr/lib/python3.6/dist-packages/graphsurgeon/node_manipulation.py", line 169, in create_plugin_node
    return create_node(name, op, trt_plugin=True, **kwargs)
  File "/usr/lib/python3.6/dist-packages/graphsurgeon/node_manipulation.py", line 152, in create_node
    return update_node(node, name, op, trt_plugin, **kwargs)
  File "/usr/lib/python3.6/dist-packages/graphsurgeon/node_manipulation.py", line 119, in update_node
    raise TypeError("Only lists of floats or ints are currently supported.")
TypeError: Only lists of floats or ints are currently supported.
```
Obviously you are right. Even if I modify featureMapShapes, the conversion fails due to "gridanchorplugin".It seems that I can only build the gridAnchorPlugin by myself or use square grid anchors.

jch-wang, 2020-10-31:
> In fact, the "gridAnchorPlugin" provided by NVIDIA probably does not support non-square grid anchors by default. In order to make it work, I think you'd need to:

I use the square grid anchors:512*512. And for input size 512 * 512, I also modified it：
`featureMapShapes=[32, 16, 8, 4, 2, 1]`
But here's the same message：
```
...............
[TensorRT] INFO: Detected 1 inputs and 2 output network tensors.
#assertionnmsPlugin.cpp,246
./build_engines.sh: line 5: 18172 Aborted                 (core dumped) python3 build_engine.py ${model}
```
I don't know what happened。

jkjung-avt, 2020-10-31:
https://github.com/NVIDIA/TensorRT/blob/release/7.1/plugin/nmsPlugin/nmsPlugin.cpp#L246

```
    // Verify C1
    ASSERT(numPriors * numLocClasses * nbBoxCoordinates == inputDims[param.inputOrder[0]].d[0]);
```

This assertion is checking whether the dimension of input 0 (bbox location data) is as expected.

For your custom model, you should double check:

1. `num_classes` should be 8 (7 custom classes plus "background"),
2. `inputOrder`.  Refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/31#issuecomment-570571648

https://github.com/jkjung-avt/tensorrt_demos/blob/e039e0824876d46443aa19e9ad7cf8f7723c713e/ssd/build_engine.py#L60-L71

jch-wang, 2020-10-31:
> For your custom model, you should double check:
> 1. `num_classes` should be 8 (7 custom classes plus "background"),
> 2. `inputOrder`.  Refer to: [#31 (comment)](https://github.com/jkjung-avt/tensorrt_demos/issues/31#issuecomment-570571648)

I changed `num_classes`  to 8. I  successfully converted & run by changing 'input_order': [1, 0, 2], to 'input_order': [0, 2, 1].
Although it seems that my model is training poorly, the problem is solved and it can run, which makes everything look great！
Thank you very much for your guidance！！！！

closed 2020-10-31 by jch-wang