jkjung-avt/tensorrt_demos #273: failure to compile plugins

Yaoxingtian, 2020-10-28:
Hi, when I make plugins , here come up with soem errors as below. Does anybody know what's the problems , I use 1080ti,py3.
thanks a lot !

errors:
...
yolo_layer.cu(151): error: identifier "PluginTensorDesc" is undefined

yolo_layer.cu(151): error: identifier "PluginTensorDesc" is undefined

yolo_layer.cu(181): error: identifier "IPluginV2IOExt" is undefined

jkjung-avt, 2020-10-28:
https://github.com/jkjung-avt/tensorrt_demos/blob/e039e0824876d46443aa19e9ad7cf8f7723c713e/plugins/Makefile#L30-L31

Please modify these paths to point to where your TensorRT include files and library files could be located.

jkjung-avt, 2020-11-04:
Closing this since there's no further question.

closed 2020-11-04 by jkjung-avt