jkjung-avt/tensorrt_demos #274: Calclating mAP on Custom Dataset

sharoseali, 2020-10-29:
HI, @jkjung-avt  I trained the model on a custom dataset and converted it to trt format for optimization. Now I want to check the mAP on my validation set. The way provided by you for mAP calculation is using py-coco tools and it calculates mAP from coco JSON format. I convert my validation set to coco JSON format but at some point, I get the error and stuck. Any information ??

`loading annotations into memory...
Done (t=0.34s)
creating index...
index created!
Loading and preparing results...
Traceback (most recent call last):
  File "eval_yolo.py", line 115, in <module>
    main()
  File "eval_yolo.py", line 105, in main
    cocoDt = cocoGt.loadRes(results_file)
  File "/home/xavier/.local/lib/python3.6/site-packages/pycocotools/coco.py", line 327, in loadRes
    'Results do not correspond to current coco set'
AssertionError: Results do not correspond to current coco set`

jkjung-avt, 2020-10-30:
> I convert my validation set to coco JSON format

This is the right way.  You just need to debug your own code.

sharoseali, 2020-10-30:
> > I convert my validation set to coco JSON format
> 
> This is the right way. You just need to debug your own code.

I tried possible options, changing my file names but still its same error. It's a problem in `cocoDt = cocoGt.loadRes(results_file)` line.
This script is using COCO API / py-coco tools and I think it's just available for evaluating on the COCO benchmark, not on custom data.   Right ??

jkjung-avt, 2020-10-31:
> This script is using COCO API / py-coco tools and I think it's just available for evaluating on the COCO benchmark, not on custom data. Right ??

No.  I do use "pycocotools" to evaluate my own custom datasets and object detection models.  You need to make sure your converted json file is correct.

sharoseali, 2020-11-01:
@jkjung-avt  Can you please share any script or tool that you use to convert **darknet Yolo forma**t data to convert **JSON COCO** format? I'll be very thankful.

jkjung-avt, 2020-11-01:
Sorry I cannot share the code.  It was developed by my colleague at work.

closed 2020-11-09 by jkjung-avt

Duarte-Nunes, 2021-04-22:
Sorry to bring up this closed post from the dead, but im getting the same problem with testing on a custom dataset. It seems that the Json file that is stored does not work and i dont understand why, im using cvat to transform the annotations on my custom dataset do Json.

What program do you use to transform?

Duarte-Nunes, 2021-04-22:
Nvm i understand it now, when you use a custom dataset the class names are stored as numbers 0 , 1 , 2 etc so i have to save de the Json file with the classes named like that aswell.

Tagging people so that they see the answer:
@sharoseali 
@jkjung-avt 
@brianegge 
@satyajitghana