jkjung-avt/tensorrt_demos #275: demo5 yolov4 has no result  QAQ, anyone help me out?

hellosummer8850, 2020-10-30:
Hi, I just wana try the demo5 Yolov4 trt to infer a single dog.jpg with trt_yolo.py,  as I followed all the steps and it seems that a trt engine is well generated, by when I try to run it, it just showed a graph but without bbx drawing, my tensorRT is with version 7.1.3.4 and cudnn 8 + cuda 11, can anyone help me out?

hellosummer8850, 2020-10-30:
my yolo folder is like this:
[image]
but no drawing of bbx like this:
[image]

closed 2020-11-01 by hellosummer8850

jkjung-avt, 2020-11-01:
https://github.com/jkjung-avt/tensorrt_demos/issues/261#issuecomment-720032042