jkjung-avt/tensorrt_demos #276: How to run yolo sample using multithreading ?

LK007CX, 2020-11-02:
It will run  well when I run `python3 trt_yolo.py --usb 0 -m yolov4-tiny-416` in command line.
But when I try to rewrite the *trt_yolo.py* with threading.thread, an error will occur.
 Here is the code I've rewritten.
How to run yolo sample using multithreading  corrently?
I hope your answer.
Your sincerely.

```python
import numpy as np
import argparse
from threading import Thread

import pycuda.autoinit  # This is needed for initializing CUDA driver

from utils.yolo_classes import get_cls_dict
from utils.camera import add_camera_args, Camera
from utils.display import open_window, set_display, show_fps
from utils.visualization import BBoxVisualization
from utils.yolo_with_plugins import TrtYOLO

WINDOW_NAME = 'TrtYOLODemo'

class DetectTensorRT(Thread):

    def __init__(self, args, parent=None):
        super(DetectTensorRT, self).__init__(parent)
        # Thread.__init__(self)
        self.args = args
        self.cam = None
        self.trt_yolo = None
        self.conf_th = 0.3
        self.vis = None
        
    
    def load_tensorRT(self):
        if self.args.category_num <= 0:
            raise SystemExit('ERROR: bad category_num (%d)!' % self.args.category_num)
        if not os.path.isfile('yolo/%s.trt' % self.args.model):
            raise SystemExit('ERROR: file (yolo/%s.trt) not found!' % self.args.model)

        self.cam = Camera(self.args)
        if not self.cam.isOpened():
            raise SystemExit('ERROR: failed to open camera!')

        cls_dict = get_cls_dict(self.args.category_num)
        yolo_dim = self.args.model.split('-')[-1]
        if 'x' in yolo_dim:
            dim_split = yolo_dim.split('x')
            if len(dim_split) != 2:
                raise SystemExit('ERROR: bad yolo_dim (%s)!' % yolo_dim)
            w, h = int(dim_split[0]), int(dim_split[1])
        else:
            h = w = int(yolo_dim)
        if h % 32 != 0 or w % 32 != 0:
            raise SystemExit('ERROR: bad yolo_dim (%s)!' % yolo_dim)

        self.trt_yolo = TrtYOLO(self.args.model, (h, w), self.args.category_num)
        self.vis = BBoxVisualization(cls_dict)

    def loop_and_detect(self):
        full_scrn = False
        fps = 0.0
        tic = time.time()
        while True:
            if cv2.getWindowProperty(WINDOW_NAME, 0) < 0:
                break
            img = self.cam.read()
            if img is None:
                break
            boxes, confs, clss = self.trt_yolo.detect(img, self.conf_th)
            img = self.vis.draw_bboxes(img, boxes, confs, clss)
            img = show_fps(img, fps)
            cv2.imshow(WINDOW_NAME, img)
            # self.image_Signal.emit(img)
            toc = time.time()
            curr_fps = 1.0 / (toc - tic)
            # calculate an exponentially decaying average of fps number
            fps = curr_fps if fps == 0.0 else (fps*0.95 + curr_fps*0.05)
            tic = toc
            key = cv2.waitKey(1)
            if key == 27:  # ESC key: quit program
                break
            elif key == ord('F') or key == ord('f'):  # Toggle fullscreen
                full_scrn = not full_scrn
                set_display(WINDOW_NAME, full_scrn)

    def run(self):
        try:
            self.load_tensorRT()
            self.loop_and_detect()
        finally:
            self.cam.release()

class ArgsHelper:

    __slots__ = 'image', 'video', 'video_looping', 'rtsp', 'rtsp_latency', 'usb', 'onboard', 'copy_frame', 'do_resize', 'width', 'height', 'category_num', 'model'
    
    def __init__(self, *args, **kwargs):
        for k, v in kwargs.items():
            setattr(self, k, v)
         

if __name__ == "__main__":
    args = {'image': None, 'video': None, 'video_looping': False, 'rtsp': None, 'rtsp_latency': 200, 'usb': 0, 'onboard': None, 'copy_frame': False, 'do_resize': False, 'width': 640, 'height': 480, 'category_num': 80, 'model': 'yolov4-tiny-416'}
    helper = ArgsHelper(image=None, video=None, video_looping=False,rtsp=None, rtsp_latency=200, usb=0, onboard=None, copy_frame=False, do_resize=False, width=640, height=480, category_num=80, model='yolov4-tiny-416')
    print(helper.model)

if __name__ == "__main__":
    args = ArgsHelper(image=None, video=None, video_looping=False,rtsp=None, rtsp_latency=200, usb=0, onboard=None, copy_frame=False, do_resize=False, width=640, height=480, category_num=80, model='yolov4-tiny-416')
    thread1 = DetectTensorRT(args)
    thread1.start()
```

jkjung-avt, 2020-11-02:
Replace this line:

```
        self.trt_yolo = TrtYOLO(self.args.model, (h, w), self.args.category_num)
```

with this:

```
        self.trt_yolo = TrtYOLO(self.args.model, (h, w), self.args.category_num, cuda_ctx=pycuda.autoinit.context)
```

since you are doing TensorRT inferencing in a child thread.

jkjung-avt, 2020-11-09:
I assume this issue has been resolved.

closed 2020-11-09 by jkjung-avt