jkjung-avt/tensorrt_demos #277: GoogLeNet questions

krambriw, 2020-11-05:
Hi, I am running your examples since long on my Jetson Nano's, working very well (using YOLOv4, full version)

Now I started to play a bit with the other examples, SSD and GoogLeNet. Basically I got it working following your instructions, no problems at all, but I wonder a few things:

1) Is there another model available for GoogLeNet that is better/able to detect people/pedestrians? The model synset_words.txt has a lot of other stuff but is not really suitable to detect humans (ok baseball player I found)
2) Does GoogLeNet provide data for bounding boxes?
3) If so, do you have a sample python code snippet how to extract and use it from the "out" result (out = net.forward(crop[None])) ?

Best regards, Walter

jkjung-avt, 2020-11-05:
GoogLeNet is a image classification model.  It only produce 1 "class" output (0~999) for the whole image.

For detecting humans, please refer to my [yolov4_crowdhuman](https://github.com/jkjung-avt/yolov4_crowdhuman).  I have included descriptions about [Deploying onto Jetson Nano](https://github.com/jkjung-avt/yolov4_crowdhuman#deploying-onto-jetson-nano) in the README.

krambriw, 2020-11-05:
Thank you ver y much for the explanation & links! I am actually very happy with YOLOV, it's really good, so this was just a test of the others. I will with high interest look into the links you provided
Best regards, Walter

closed 2020-11-05 by krambriw