jkjung-avt/tensorrt_demos #279: Request for a DeepSort demo

jamesee, 2020-11-09:
Hi jkjung

Thank you for the superb demos. I can now run Yolov4-tiny on tensorRT at more than 50fps on jetson nx.

May I request a deepsort demo using your plugin. Many thanks in advance.

James

jkjung-avt, 2020-11-09:
@jamesee  Thanks for the suggestion.  But I don't have a plan to add DeepSort (or other trackers) into the demos in this repository.  I will consider it when I have the time and energy to create a new project.

jamesee, 2020-11-09:
@jkjung-avt 

Noted with thanks.

GeekAlexis, 2020-11-10:
@jamesee I think I have exactly what you want: https://github.com/GeekAlexis/FastMOT 
@jkjung-avt Thanks for your great repo

jkjung-avt, 2020-11-10:
@GeekAlexis  Sounds good.  Thanks for sharing.

jamesee, 2020-11-12:
> @jamesee Here is my implementation: https://github.com/GeekAlexis/FastMOT
> @jkjung-avt Thanks for your great repo

@GeekAlexis 
Thank you for the info and great repo. Will study in detail.

closed 2020-11-12 by jkjung-avt