jkjung-avt/tensorrt_demos #280: How to run     mtcnn.detect(img, minsize=minsize)    using a  child thread?

17120033, 2020-11-10:
Hi @jkjung-avt,

now i need to do  mtcnn.detect()  in  the while loop  using a child thread, 
there comes CUDA context problem,
I refer to the solution of multithreading in trtyolo,  like #276,
but I find that trt_mtcnn and trt_yolo are not very similar. 
How should I solve this problem?
hope your answer!
Your sincerely!

17120033, 2020-11-10:
the cuda context was initialized in the main thread (mtcnn=TrtMtcnn(engine))  while the inference (mtcnn.detect()) was in a child thread.

jkjung-avt, 2020-11-12:
Please post the exact error messages (as detailed as possible) you're seeing.  I think it could be a different problem.

jkjung-avt, 2020-11-19:
Closing this due to no response.

closed 2020-11-19 by jkjung-avt

17120033, 2020-11-19:
Sorry,  i forgot to tell you @jkjung-avt,  We have noticed that the error is due to our code, and thank you very much again for your excellent work！！！！！！！！！！！！！！

jkjung-avt, 2020-11-19:
@17120033  Good to hear that.