jkjung-avt/tensorrt_demos #281: yolov3 output is zero

Daniel-1597, 2020-11-12:
[image]
I have gone through issues but I couldn't find the reason please let me know what can be the issue.

Thanks, 
Daniel

jkjung-avt, 2020-11-12:
Your issue description is too unspecific, so I cannot help much.

Are you running the code with the downloaded (pretrained with COCO) yolov3 model?

Otherwise, check out:

* https://github.com/jkjung-avt/tensorrt_demos/issues/275
* https://github.com/jkjung-avt/tensorrt_demos/issues/261

Daniel-1597, 2020-11-12:
I have downloaded the weights from.sh file after that I have converted to onnx model and trt while I am doing inference the model is not detecting anything.

Daniel-1597, 2020-11-12:
I am trying it on p2.8xlarge with cuda 10.0 and trt 6.0.1.5

jkjung-avt, 2020-11-12:
Try updating the CUDA driver.

Daniel-1597, 2020-11-12:
I will update and I will check it

Daniel-1597, 2020-11-12:
Configuration on Jetson AGX
CUDA-10.2
TensorRT - 7.1.3.0

I have tried it on Jetson AGX it worked with coco weights... Now when I am trying with my own custom yolov3 model.. I am getting  an error 
**Weights are outside of the fp16 range. A possible fix is to retrain the model with regularization to bring the magnitude of the weights down.)
ERROR: Failed to build the TensorRT engine!**
Can you please tell me How to train the yolov3 model with FP16? I am unable to train I tried multiple methods. 
Thanks, 
Daniel

jkjung-avt, 2020-11-12:
https://github.com/jkjung-avt/tensorrt_demos/issues/161#issuecomment-662473343

Daniel-1597, 2020-11-12:
Ok thanks I have one more question regarding int8 conversion I am getting out of memory error.

jkjung-avt, 2020-11-13:
Please refer to the FPS numbers I posted in the README (for Jetson Nano).  You would be able to achieve higher FPS if you code with C++ (or use DeepStream SDK).

Reference: https://github.com/AlexeyAB/darknet/pull/5453#issuecomment-663593512

Daniel-1597, 2020-11-14:
One more question I trained my custom model by changing the weight decay still there is the same problem weights are out of fp16 range... can you suggest weight decay value or how to train the model so that weights will be in fp16

jkjung-avt, 2020-11-14:
You could refer to my [yolov4_crowdhuman examples](https://github.com/jkjung-avt/yolov4_crowdhuman/tree/master/cfg).  In addition to that, I have also trained other custom models in a very similar way.  I always used `decay=0.0005` as set in the original [yolov4-custom.cfg](https://github.com/AlexeyAB/darknet/blob/master/cfg/yolov4-custom.cfg) and [yolov4-tiny-custom.cfg](https://github.com/AlexeyAB/darknet/blob/master/cfg/yolov4-tiny-custom.cfg).  And the resulting model could be optimized as TensorRT FP16 engines without any problem.

I think 0.05 is too large.

Daniel-1597, 2020-11-14:
yeah I will go through them if I have any doubt I will let you know. Thanks for your immediate replies :-)

Daniel-1597, 2020-11-14:
I will attach my weight files and cfg can you try converting them into TRT

jkjung-avt, 2020-11-14:
Please work on that by yourself.

Daniel-1597, 2020-11-14:
sure

closed 2020-11-16 by jkjung-avt