jkjung-avt/tensorrt_demos #282: Output shape of the TRT model

Daniel-1597, 2020-11-19:
Can you explain me why is the output shape of the model is 56784 after converting the model to Tensorrt the last layer output of the onnx model is (30,52,52) which is converted to (56784,1,1).

closed 2020-11-20 by Daniel-1597

jkjung-avt, 2020-11-20:
The output shape of my "yolo_layer" plugin is defined here: https://github.com/jkjung-avt/tensorrt_demos/blob/26fb5a441134b0a58bd5804a5c7cdd1183a14554/plugins/yolo_layer.cu#L119-L120

where `Detection` is a struct of 7 floats as defined here: https://github.com/jkjung-avt/tensorrt_demos/blob/26fb5a441134b0a58bd5804a5c7cdd1183a14554/plugins/yolo_layer.h#L28-L33

So for a yolo_layer with a 1x30x52x52 input tensor (number_of_classes = 5 and number_of_anchors = 3), the output shape would be  1x(52\*52\*3\*7)x1x1 or 1x56784x1x1.

Daniel-1597, 2020-11-20:
Yeah i figured it out and that's why I closed it...i have converted my custom model into TRT and I. Getting around 42 FPS....can we achieve more than that.

jkjung-avt, 2020-11-21:
https://github.com/jkjung-avt/tensorrt_demos/issues/200#issuecomment-707460415

Daniel-1597, 2020-11-21:
Thanks for your response 😊