jkjung-avt/tensorrt_demos #288: assert size == grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size AssertionError

zabir-nabil, 2020-12-01:
I trained a yolov4-tiny with 8 classes, ran `yolo_to_onnx`, and `onnx_to_tensorrt` and created the .trt file. 

I modifed the `trt_yolo` script to work with single image, but getting the following error: 
```
Traceback (most recent call last):
  File "/home/nabil/project/tensorrt_demos/utils/yolo_with_plugins.py", line 258, in __init__
    allocate_buffers(self.engine, grid_sizes)
  File "/home/nabil/project/tensorrt_demos/utils/yolo_with_plugins.py", line 169, in allocate_buffers
    assert size == grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size
AssertionError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "trt_yolo_img.py", line 114, in <module>
    main()
  File "trt_yolo_img.py", line 102, in main
    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num)
  File "/home/nabil/project/tensorrt_demos/utils/yolo_with_plugins.py", line 260, in __init__
    raise RuntimeError('fail to allocate CUDA resources') from e
RuntimeError: fail to allocate CUDA resources
Exception ignored in: <bound method TrtYOLO.__del__ of <utils.yolo_with_plugins.TrtYOLO object at 0x7f885df780>>
Traceback (most recent call last):
  File "/home/nabil/project/tensorrt_demos/utils/yolo_with_plugins.py", line 267, in __del__
    del self.outputs
AttributeError: outputs

```

jkjung-avt, 2020-12-01:
`    assert size == grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size`

Please print out the values of `size` and `grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size`, so that I could make better guess about what the problem could be.

zabir-nabil, 2020-12-01:
```
size 3549
grid suze [2704, 676, 169]
output idx 0
engine_max_size 1
```
This is what I get from print.

zabir-nabil, 2020-12-01:
This is my .cfg file

```
[net]
# Testing
#batch=1
#subdivisions=1
# Training
batch=128
subdivisions=16
width=416
height=416
channels=3
momentum=0.9
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.00261
burn_in=1000
max_batches = 20000
# max_batches = 500200
policy=steps
steps=16000,18000
scales=.1,.1

[convolutional]
batch_normalize=1
filters=32
size=3
stride=2
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=64
size=3
stride=2
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=64
size=3
stride=1
pad=1
activation=leaky

[route]
layers=-1
groups=2
group_id=1

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=leaky

[route]
layers = -1,-2

[convolutional]
batch_normalize=1
filters=64
size=1
stride=1
pad=1
activation=leaky

[route]
layers = -6,-1

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=128
size=3
stride=1
pad=1
activation=leaky

[route]
layers=-1
groups=2
group_id=1

[convolutional]
batch_normalize=1
filters=64
size=3
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=64
size=3
stride=1
pad=1
activation=leaky

[route]
layers = -1,-2

[convolutional]
batch_normalize=1
filters=128
size=1
stride=1
pad=1
activation=leaky

[route]
layers = -6,-1

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=256
size=3
stride=1
pad=1
activation=leaky

[route]
layers=-1
groups=2
group_id=1

[convolutional]
batch_normalize=1
filters=128
size=3
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=128
size=3
stride=1
pad=1
activation=leaky

[route]
layers = -1,-2

[convolutional]
batch_normalize=1
filters=256
size=1
stride=1
pad=1
activation=leaky

[route]
layers = -6,-1

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=512
size=3
stride=1
pad=1
activation=leaky

##################################

[convolutional]
batch_normalize=1
filters=256
size=1
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=512
size=3
stride=1
pad=1
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=39
activation=linear

[yolo]
mask = 3,4,5
anchors = 10,14,  23,27,  37,58,  81,82,  135,169,  344,319
classes=8
num=6
jitter=.3
scale_x_y = 1.05
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
ignore_thresh = .7
truth_thresh = 1
random=0
resize=1.5
nms_kind=greedynms
beta_nms=0.6

[route]
layers = -4

[convolutional]
batch_normalize=1
filters=128
size=1
stride=1
pad=1
activation=leaky

[upsample]
stride=2

[route]
layers = -1, 23

[convolutional]
batch_normalize=1
filters=256
size=3
stride=1
pad=1
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=39
activation=linear

[yolo]
mask = 1,2,3
anchors = 10,14,  23,27,  37,58,  81,82,  135,169,  344,319
classes=8
num=6
jitter=.3
scale_x_y = 1.05
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
ignore_thresh = .7
truth_thresh = 1
random=1
resize=1.5
nms_kind=greedynms
beta_nms=0.6

```

jkjung-avt, 2020-12-01:
As I have described in [TensorRT YOLOv3 For Custom Trained Models](https://jkjung-avt.github.io/trt-yolov3-custom/), please name your cfg and weights files as "yolov4-tiny-XXXXXX.cfg" and "yolov4-tiny-XXXXXX.weights".  My code relies on the file names to determine model architecture and output sizes.  (You should re-run yolo_to_onnx and onnx_to_tensorrt after modifying the file names.)

https://github.com/jkjung-avt/tensorrt_demos/blob/26fb5a441134b0a58bd5804a5c7cdd1183a14554/utils/yolo_with_plugins.py#L215-L228

zabir-nabil, 2020-12-02:
Now, I'm getting the following error when running `onnx_to_tensorrt`

```
 python3 onnx_to_tensorrt.py -m yolov4-lpv-416 --category_num 8
Loading the ONNX file...
Adding yolo_layer plugins...
debug
size: [[52, 52], [26, 26], [13, 13]]
anchors: [[81, 82, 135, 169, 344, 319], [23, 27, 37, 58, 81, 82]]
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 195, in <module>
    main()
  File "onnx_to_tensorrt.py", line 184, in main
    args.model, args.category_num, args.int8, args.dla_core, args.verbose)
  File "onnx_to_tensorrt.py", line 113, in build_engine
    network, model_name, category_num, TRT_LOGGER)
  File "/home/nabil/project/tensorrt_demos/yolo/plugins.py", line 123, in add_yolo_plugins
    (len(anchors), len(yolo_whs)))
ValueError: bad number of yolo layers: 2 vs. 3

```

closed 2020-12-02 by zabir-nabil