jkjung-avt/tensorrt_demos #289: calibration

alicera, 2020-12-05:
I want to generate a calibration file for yolov4.
Do you have any idea？

jkjung-avt, 2020-12-05:
Refer to [Demo #6: Using INT8 and DLA core](https://github.com/jkjung-avt/tensorrt_demos#demo-6-using-int8-and-dla-core).  In step 2, when you call `python3 onnx_to_tensorrt.py -v --int8  -m yolov4-int8-608`, a calibration cache file would also be generated: "calib_yolov4-int8-608.bin".

alicera, 2020-12-05:
Can I only use  the  onnx model to generate .bin ？
But the onnx_to_tensorrt.py seem to use the .cfg

jkjung-avt, 2020-12-05:
1. You need to prepare image files, as well as the ONNX file, to generate the calibration cache file (if you use my "onnx_to_tensorrt.py" script).

2. As I already described in the README, please follow steps in [Demo #5: YOLOv4](https://github.com/jkjung-avt/tensorrt_demos#demo-5-yolov4) to generate the ONNX files first.

alicera, 2020-12-05:
COMMAND:
python3 onnx_to_tensorrt.py -v --int8 -m yolov4-416
I have the yolov4-416.onnx and I take the yolov4-416.cfg out

ERROR:
Adding yolo_layer plugins...
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 195, in <module>
    main()
  File "onnx_to_tensorrt.py", line 184, in main
    args.model, args.category_num, args.int8, args.dla_core, args.verbose)
  File "onnx_to_tensorrt.py", line 113, in build_engine
    network, model_name, category_num, TRT_LOGGER)
  File "/host/home/sa/Music/TRT/tensorrt_demos/yolo/plugins.py", line 112, in add_yolo_plugins
    if not verify_classes(model_name, num_classes):
  File "/host/home/sa/Music/TRT/tensorrt_demos/yolo/plugins.py", line 54, in verify_classes
    with open(cfg_file_path, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'yolov4-416.cfg'

jkjung-avt, 2020-12-06:
That's right.  My "onnx_to_tensorrt.py" script would look for the cfg file in order to determine the correct shapes/parameters for the yolo_layer plugin.

closed 2020-12-06 by alicera