jkjung-avt/tensorrt_demos #291: Question for conversion

jahwanoh, 2020-12-07:
Hi,
Thank you for your contribution..
I have question regarding conversion. 
should I convert the model on target platform?
for example, If I'm converting yolo.weight to ONNX (or trt), I'm converting the model on other linux machine. after converting I copy this converted model to Jetson AGX Xavier and use for detection.
does it make sense? or should I convert the model on Jetson from the beginning?

jkjung-avt, 2020-12-07:
"onnx_to_tensorrt.py" (also including compiling the yolo_layer plugin) must be done on the target platform.

"yolo_to_onnx.py" could be done on other machines.  For example, it's OK to do this step on a x86_64 PC and then copy the onnx file to your Jetson AGX Xavier to do TensorRT optimization and inferencing.

jahwanoh, 2020-12-09:
Make sense. Thx!!

closed 2020-12-09 by jahwanoh