jkjung-avt/tensorrt_demos #292: Custom Yolo Detection - Unable to replicate original model outputs

Terizian, 2020-12-08:
First of all, thank you so much for putting this repository together. It was one of the most straightforward to work with for my case.

I have two yolov3 models that I've trained with custom data and have followed the yolov3/v4 tutorial in your README page. I've successfully converted both models into trt engines. However, upon detection there are no boxes drawn. In fact, the output is all zeros. 

`[array([0., 0., 0., ..., 0., 0., 0.], dtype=float32), array([0., 0., 0., ..., 0., 0., 0.], dtype=float32), array([0., 0., 0., ..., 0., 0., 0.], dtype=float32)]`

I have tested the original models and they show have successfully detected objects in the sample images I have. Could you guide me as to what to check in order to debug this issue? Also, please let me know if you require further information.

Terizian, 2020-12-08:
As a matter of fact, I follow the demo (with the downloaded yolo models) and the dog example was not showing any boxes either. 
TensorRT version is 7.1.3.4 and ONNX version is 1.4.1 using Python 3.6.9

jkjung-avt, 2020-12-08:
I assume you are testing the code on an x86_64 PC.  In this case, please try updating your NVIDIA driver first.

Reference: https://github.com/jkjung-avt/tensorrt_demos/issues/261#issuecomment-719520590

Terizian, 2020-12-08:
> I assume you are testing the code on an x86_64 PC. In this case, please try updating your NVIDIA driver first.
> 
> Reference: [#261 (comment)](https://github.com/jkjung-avt/tensorrt_demos/issues/261#issuecomment-719520590)

Here is the information displayed from running nvidia-smi: 
NVIDIA-SMI 455.38       Driver Version: 455.38       CUDA Version: 11.1

jkjung-avt, 2020-12-08:
Please also make sure you have a compatible version of cuDNN installed on your system.

Otherwise, please try to run one of [official TensorRT samples](https://docs.nvidia.com/deeplearning/tensorrt/sample-support-guide/index.html) and make sure your installation is good.

Terizian, 2020-12-08:
> Please also make sure you have a compatible version of cuDNN installed on your system.
> 
> Otherwise, please try to run one of [official TensorRT samples](https://docs.nvidia.com/deeplearning/tensorrt/sample-support-guide/index.html) and make sure your installation is good.

You're right. The documentation from NVIDIA states that the TensorRT version : 
This TensorRT release supports CUDA 10.2 and CUDA 11.0 RC.

I have CUDA 11.1 installed. Will try to downgrade and report back

Terizian, 2020-12-08:
> Please also make sure you have a compatible version of cuDNN installed on your system.
> 
> Otherwise, please try to run one of [official TensorRT samples](https://docs.nvidia.com/deeplearning/tensorrt/sample-support-guide/index.html) and make sure your installation is good.

I have CUDA 10.2 installed with cuDNN 7.6.5 with the same TensorRT version but the demo didn't work. Here are the driver details: NVIDIA-SMI 440.33.01    Driver Version: 440.33.01    CUDA Version: 10.2

jkjung-avt, 2020-12-08:
Do official TensorRT samples work?

Terizian, 2020-12-09:
> Do official TensorRT samples work?

I have tried one under the "introductory_parser_samples" but it throws an error `pycuda._driver.LogicError: cuInit failed: forward compatibility was attempted on non supported HW`
I had the 2020.1 version installed. I uninstalled it and used your script to install pycuda (the one in the ssd folder) but I still get the same error. Could you please advise me?

Terizian, 2020-12-09:
> I have tried one under the "introductory_parser_samples" but it throws an error `pycuda._driver.LogicError: cuInit failed: forward compatibility was attempted on non supported HW`
> I had the 2020.1 [...]

Quick update: My PC needed a reboot, and pycuda is working fine now. I ran the **onnx_resnet50.py** script and this is the output: 
`Correctly recognized /usr/src/tensorrt/data/resnet50/binoculars.jpeg as binoculars`

I tried running your sample code one more time, and still no boxes displayed on dog.jpg.

Terizian, 2020-12-09:
Sorry for all the comments but it might be worth mentioning that I'm working with a PC that has two GeForce GTX 1080 Ti. I haven't found any information regarding TensorRT support for it. Might that be the issue?

Meanwhile, I'll try to replicate the setup on a Jetson AGX Xavier and report back.

jkjung-avt, 2020-12-09:
> Sorry for all the comments but it might be worth mentioning that I'm working with a PC that has two GeForce GTX 1080 Ti. I haven't found any information regarding TensorRT support for it. Might that be the issue?

Although I myself only tested TensorRT yolov4/yolov4-tiny engines on GTX 2080 Ti, I'm sure TensorRT also works for GTX 1080 Ti.

I'm not sure what problem you're encountering.  Hopefully, it would just work on your Jetson AGX Xavier.

Terizian, 2020-12-09:
> Although I myself only tested TensorRT yolov4/yolov4-tiny engines on GTX 2080 Ti, I'm sure TensorRT also works for GTX 1080 Ti.
>
> I'm not sure what problem you're encountering. Hopefully, it would [...]

It ran seemlessly on the Xavier, although I followed the exact same steps. I wonder if there's a support matrix available to confirm my suspicions. 
Thank you so much for all your help. I will close this thread now

closed 2020-12-09 by Terizian

jkjung-avt, 2020-12-20:
> Sorry for all the comments but it might be worth mentioning that I'm working with a PC that has two GeForce GTX 1080 Ti. I haven't found any information regarding TensorRT support for it. Might that be the issue?

By the way, could you check if you have set cuda compute to "61" (instead of 75) for GTX-1080 Ti in the Makefile?  Again, TensorRT (FP16) should work on GTX-1080 Ti.

https://github.com/jkjung-avt/tensorrt_demos/blob/15a725fee16214585ed21aebba59bccddaa4f010/plugins/Makefile#L21

Terizian, 2020-12-20:
> By the way, could you check if you have set cuda compute to "61" (instead of 75) for GTX-1080 Ti in the Makefile? Again, TensorRT (FP16) should work on GTX-1080 Ti. [...]

It worked with the compute=61! Thanks a lot for all your help!