jkjung-avt/tensorrt_demos #293: building trt file fails

bhaktatejas922, 2020-12-09:


bhaktatejas922, 2020-12-09:
I get this issue when building the trt file on the new Jetson Nano 2GB: 
>> python3 onnx_to_tensorrt.py -m yolov4-416
Loading the ONNX file...
Adding yolo_layer plugins...
Building an engine.  This would take a while...
(Use "--verbose" or "-v" to enable verbose logging.)
Killed

I think its failing due to lack of memory? I've tried it with swapfile size 6GB and 8GB, still not working

bhaktatejas922, 2020-12-09:
Also. huge thanks to @jkjung-avt and contributors for this repo. Doing the stuff Nvidia cant!

jkjung-avt, 2020-12-09:
> python3 onnx_to_tensorrt.py -m yolov4-416
> ......
> Killed

Yes.  This is very likely due to lack of memory.

Adding swap could not solve it if the problem is lacking GPU memory.  One more thing you could probably try is to reduce GPU workspace size, say from 1GB to 256MB.

https://github.com/jkjung-avt/tensorrt_demos/blob/26fb5a441134b0a58bd5804a5c7cdd1183a14554/yolo/onnx_to_tensorrt.py#L132

> Also. huge thanks to @jkjung-avt and contributors for this repo. Doing the stuff Nvidia cant!

This comment is very much appreciated.

bhaktatejas922, 2020-12-09:
I'll try that and update with results. Here's what it looks like before dying in verbose mode 

`[TensorRT] VERBOSE: Fusing PWN(105_convolutional_softplus, 105_convolutional_tanh) with 105_convolutional_mish
[TensorRT] VERBOSE: After vertical fusions: 235 layers
[TensorRT] VERBOSE: After final dead-layer removal: 235 layers
[TensorRT] VERBOSE: Merging layers: 005_convolutional || 003_convolutional
[TensorRT] VERBOSE: Merging layers: 015_convolutional || 013_convolutional
[TensorRT] VERBOSE: Merging layers: 028_convolutional || 026_convolutional
[TensorRT] VERBOSE: Merging layers: 059_convolutional || 057_convolutional
[TensorRT] VERBOSE: Merging layers: 090_convolutional || 088_convolutional
[TensorRT] VERBOSE: After tensor merging: 230 layers
[TensorRT] VERBOSE: Eliminating concatenation 010_route
[TensorRT] VERBOSE: Retargeting 009_convolutional_mish to 010_route
[TensorRT] VERBOSE: Retargeting 003_convolutional_mish to 010_route
[TensorRT] VERBOSE: Eliminating concatenation 023_route
[TensorRT] VERBOSE: Retargeting 022_convolutional_mish to 023_route
[TensorRT] VERBOSE: Retargeting 013_convolutional_mish to 023_route
[TensorRT] VERBOSE: Eliminating concatenation 054_route
[TensorRT] VERBOSE: Retargeting 053_convolutional_mish to 054_route
[TensorRT] VERBOSE: Retargeting 026_convolutional_mish to 054_route
[TensorRT] VERBOSE: Eliminating concatenation 085_route
[TensorRT] VERBOSE: Retargeting 084_convolutional_mish to 085_route
[TensorRT] VERBOSE: Retargeting 057_convolutional_mish to 085_route
[TensorRT] VERBOSE: Eliminating concatenation 104_route
[TensorRT] VERBOSE: Retargeting 103_convolutional_mish to 104_route
[TensorRT] VERBOSE: Retargeting 088_convolutional_mish to 104_route
[TensorRT] VERBOSE: Eliminating concatenation 114_route
[TensorRT] VERBOSE: Retargeting 113_maxpool to 114_route
[TensorRT] VERBOSE: Retargeting 111_maxpool to 114_route
[TensorRT] VERBOSE: Retargeting 109_maxpool to 114_route
[TensorRT] VERBOSE: Generating copy for 108_convolutional_lrelu to 114_route
[TensorRT] VERBOSE: Eliminating concatenation 122_route
[TensorRT] VERBOSE: Retargeting 121_convolutional_lrelu to 122_route
[TensorRT] VERBOSE: Generating copy for 119_upsample to 122_route
[TensorRT] VERBOSE: Eliminating concatenation 132_route
[TensorRT] VERBOSE: Retargeting 131_convolutional_lrelu to 132_route
[TensorRT] VERBOSE: Generating copy for 129_upsample to 132_route
[TensorRT] VERBOSE: Eliminating concatenation 143_route
[TensorRT] VERBOSE: Retargeting 142_convolutional_lrelu to 143_route
[TensorRT] VERBOSE: Retargeting 127_convolutional_lrelu to 143_route
[TensorRT] VERBOSE: Eliminating concatenation 154_route
[TensorRT] VERBOSE: Retargeting 153_convolutional_lrelu to 154_route
[TensorRT] VERBOSE: Retargeting 117_convolutional_lrelu to 154_route
[TensorRT] VERBOSE: After concat removal: 223 layers
[TensorRT] VERBOSE: Graph construction and optimization completed in 1.32956 seconds.
Killed
`

bhaktatejas922, 2020-12-09:
@jkjung-avt any way you could post the trt file for yolov4-416 and yolov4-608? Maybe google drive? Changed the max workspace size to 1 << 28, but still didnt work

jkjung-avt, 2020-12-09:
I don't have a Jetson Nano 2GB.  I'm not able to build the TensorRT engine files for you...

bhaktatejas922, 2020-12-09:
Ah I see. I figured the file would be the same for jetson Nano 4gb and 2gb since the hardware is the same. Are you sure it would be different?

jkjung-avt, 2020-12-09:
> I figured the file would be the same for jetson Nano 4gb and 2gb since the hardware is the same.

The available GPU memory is different.  I think TensorRT would build different engines on these 2 platforms.

bhaktatejas922, 2020-12-09:
Would you mind sharing your yolov4-416.trt file so I could try? If you have it handy. Thanks.

Also, any other tips for getting it to run? I tried running it with the hdmi monitor unplugged, I tried manually setting vm.swappiness higher (though this doesn't affect gpu men), and I tried lowering workspace size to 1 << 28, 1 << 27. None of these worked. Running out of ideas 😟

jkjung-avt, 2020-12-09:
You could try yolov4-288,  or yolov4-tiny models.

bhaktatejas922, 2020-12-09:
Failed on 288 as well. I'll try with tiny models soon. Let me know if something comes to mind for a workaround!

jkjung-avt, 2020-12-23:
> python3 onnx_to_tensorrt.py -m yolov4-416
> ......
> Killed

I've tested/verified the code repeatedly on Jetson Nano 4GB.  I think this problem is very likely due to lack of GPU memory (on Jetson Nano 2GB) and I don't have a solution to it.

I'm going to close this issue for now.

closed 2020-12-23 by jkjung-avt

huuthieu, 2021-06-18:
> @jkjung-avt any way you could post the trt file for yolov4-416 and yolov4-608? Maybe google drive? Changed the max workspace size to 1 << 28, but still didnt work

Try to change workspace size to 1<<25