jkjung-avt/tensorrt_demos #294: Resolution problem

LK007CX, 2020-12-10:
Why can't I run yolov4 tiny's USB video program at 1280 * 720 resolution?
But I seem to have run successfully before.

The terminal will report the following error.

```
/usr/bin/python3 /home/edit/code/Multichannel_Camera_Demo/main.py
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (515) startPipeline OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1055) setProperty OpenCV | GStreamer warning: no pipeline
WARNING:root:Camera: starting while cap is not opened!
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
ERROR: failed to open camera!
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pycuda-2019.1.2-py3.6-linux-aarch64.egg/pycuda/autoinit.py", line 14, in _finish_up
    context.pop()
pycuda._driver.LogicError: context::pop failed: invalid device context - cannot pop non-current context
QObject::~QObject: Timers cannot be stopped from another thread
```

LK007CX, 2020-12-10:
After several runs, 800 * 480 resolution will not work.

jkjung-avt, 2020-12-10:
> /usr/bin/python3 /home/edit/code/Multichannel_Camera_Demo/main.py

I have no idea what this code does...

> ERROR: failed to open camera!

One possible problem is that your camera does not support the requested resolution.

LK007CX, 2020-12-11:
Thanks.
I tried to change a USB camera and it worked again.
The compatibility of the camera really puzzles me.

closed 2020-12-11 by LK007CX