jkjung-avt/tensorrt_demos #295: Rtsp streaming Issue

Darshcg, 2020-12-11:
Hi @jkjung-avt,

I am trying to run the YoloV3 TRT inference through RTSP, but I am getting an issue as: 
**WARNING:root:Camera : starting while cap is not opened!
ERROR: Failed to open camera!**

I had attached the pic below.

[image]

Pinging the same IP address is working in the Jetson nano, but do no why I am facing this issue.

May I know the reason I am facing the issue, so that I will try to resolve?

Thanks in advance,
Darshan

jkjung-avt, 2020-12-11:
Have you installed the gstreamer plugins which are necessary to demux/decode RTSP H.264 streams?  And you are using a RTSP source in H.264 format, right?

Reference: [How to Capture and Display Camera Video with Python on Jetson TX2](https://jkjung-avt.github.io/tx2-camera-with-python/)

Darshcg, 2020-12-11:
Hi @jkjung-avt,
Thank you very much for your reply.
I am using this Camera: https://www.cpplusworld.com/cp-nnc-t451-t which has **H.265, H.264, MJPEG dual-stream encoding**

Darshcg, 2020-12-11:
Hi @jkjung-avt,
I have installed gstreamer plugins and checked, but still the same issue.

jkjung-avt, 2020-12-11:
If you are not using a H.264 stream, you need to modify the "open_cam_rtsp()" function by yourself, e.g. use a H.265 decoder in the GStreamer pipeline instead.

https://github.com/jkjung-avt/tensorrt_demos/blob/26fb5a441134b0a58bd5804a5c7cdd1183a14554/utils/camera.py#L53-L63

Darshcg, 2020-12-14:
Hi @jkjung-avt,
I configured my IP Camera with H.264 Streaming, It is working Fine. 

Thank you very much!

closed 2020-12-14 by Darshcg