jkjung-avt/tensorrt_demos #297: Multiple face finding problem on video in MTCNN Demo

ssefadursun, 2020-12-17:
Hello,

When I test the code trt_mtcnn.py on images in the MTCNN demo, there is no problem in finding multiple faces in the images.
But when I run the same code on the video or webcam, it only finds one face in the same frame, it cannot find more than one face at the same time. ( I run the tests on Xavier NX (JetPack-4.4) )

When I made adjustments on the detection confidence score threshold parameter, the problem of finding multiple faces on the same frame continued and the False Positive Face ratio also increased.

Does anyone have any idea what I can try? 
Thank you in advance.

jkjung-avt, 2020-12-25:
I could reproduce the problem with a USB webcam, but I don't know what the problem might be...

hubalaga, 2021-01-05:
I have the same problem. Jetson Nano, Jetpack 4.4, webcam.

hubalaga, 2021-01-05:
If I run this example and the camera at start captures several faces at once, it works beautifully. If we leave the room and goes back she only catches one face and alternately - me and other.

BTW: I noticed that facial recognition works better if I run my webcam at maximum resolution (in my case it is width = (int) 3264, height = (int) 2464), when I add interpolation-method = 5 when invoking it and run this example for values around 700 pixels wide.

So in utils\camera.py i change this line:

```
    elif 'nvarguscamerasrc' in gst_elements:
        gst_str = ('nvarguscamerasrc ! '
                   'video/x-raw(memory:NVMM), '
                   'width=(int)3264, height=(int)2464, '
                   'format=(string)NV12, framerate=(fraction)21/1 ! '
                   'nvvidconv interpolation-method=5 flip-method=2 ! '
                   'video/x-raw, width=(int){}, height=(int){}, '
                   'format=(string)BGRx ! '
                   'videoconvert ! appsink').format(width, height)
```

but that doesn't solve the main problem unfortunately :-(

c7934597, 2021-01-20:
Hi all, this problem has answer?

hubalaga, 2021-02-15:
I'm afraid not .. I'm waiting too :-)

jkjung-avt, 2021-02-17:
I'm consolidating this issue with https://github.com/jkjung-avt/tensorrt_demos/issues/313.  Please track the issue there.

closed 2021-02-17 by jkjung-avt