jkjung-avt/tensorrt_demos #298: This repo packaged by  module "pyinstaller"  got  a  executed  file,  then  it  can not  be executed?

Alphonses, 2020-12-18:
@Jkjung ,thank you very much for your efforts in this Repo.I benefited a  lot.

I can  run this repo  with the  source code  of  "trt_yolo.py".But  i  packaged this repo  by  module "pyinstaller"  got  a  executed  file,  then  it  can not  work. it let me confused.  Can you  give me  some ideas?   Thank  a  lot!!   

error  log   is :

File "utils/yolo_with_plugins.py", line 255, in __init__
    self.context = self.engine.create_execution_context()
AttributeError: 'NoneType' object has no attribute 'create_execution_context'

why  engine  is   'NoneType' object.  I can ensure   the engine is  loaded  by  funtion "_load_engine"!!

def _load_engine(self):
        TRTbin = 'yolo/%s.trt' % self.model
        with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:
            return runtime.deserialize_cuda_engine(f.read())

The error  log  As described below:

1. packaged this repo  by  module "pyinstaller"  got  a  executed  file

pyinstaller -F --hidden-import=cv2 --hidden-import=onnx.onnx_cpp2py_export  --hidden-import=tensorrt --paths="/usr/local/lib/python3.6/dist-packages/cv2/python-3.6"   trt_yolo.py

2.run  the  executed  file  "trt_yolo_app"

hnsd@hnsd-desktop:~/Projects/tensorrt_demos$ ./trt_yolo_app 
onnx_file is  exist !!
engine_file is  exist !!
[TensorRT] ERROR: /home/jenkins/workspace/TensorRT/helpers/rel-7.1/L1_Nightly_Internal/build/source/rtSafe/resources.h (460) - Cuda Error in loadKernel: 3 (initialization error)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
Traceback (most recent call last):
  File "utils/yolo_with_plugins.py", line 255, in __init__
    self.context = self.engine.create_execution_context()
AttributeError: 'NoneType' object has no attribute 'create_execution_context'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "app_yolov4.py", line 141, in <module>
    test()
  File "app_yolov4.py", line 20, in test
    trt_yolov4 = load_yolov4_model(detection_id, width)
  File "port_trt_yolov4.py", line 80, in load_yolov4_model
    trt_yolov4 = TrtYOLO(detection_id, (width, width),  class_num, cuda_ctx=None)  # 416 or 608
  File "utils/yolo_with_plugins.py", line 259, in __init__
    raise RuntimeError('fail to allocate CUDA resources') from e
RuntimeError: fail to allocate CUDA resources
[15105] Failed to execute script app_yolov4
Exception ignored in: <bound method TrtYOLO.__del__ of <utils.yolo_with_plugins.TrtYOLO object at 0x7f7c988668>>
Traceback (most recent call last):
  File "utils/yolo_with_plugins.py", line 266, in __del__
    del self.outputs
AttributeError: outputs

jkjung-avt, 2020-12-18:
> [TensorRT] ERROR: /home/jenkins/workspace/TensorRT/helpers/rel-7.1/L1_Nightly_Internal/build/source/rtSafe/resources.h (460) - Cuda Error in loadKernel: 3 (initialization error)

Fix this error first.

Referring to [CUDA error codes](https://docs.nvidia.com/cuda/cuda-runtime-api/group__CUDART__TYPES.html#group__CUDART__TYPES_1g3f51e3575c2178246db0a94a430e0038), error code 3 actually means "The API call failed because the CUDA driver and runtime could not be initialized."

I'd guess this CUDA initialization failure (i.e. TensorRT initialization failure) is caused by missing library files.  Try to find those missing .so files and add them into pyinstaller spec.

jkjung-avt, 2020-12-25:
Has this issue been resolved?  Could I close the issue?

Alphonses, 2020-12-25:
Thank you for your reply. I fix the error  by add pycuda into pyinstaller spec:[TensorRT] ERROR: /home/jenkins/workspace/TensorRT/helpers/rel-7.1/L1_Nightly_Internal/build/source/rtSafe/resources.h (460) - Cuda Error in loadKernel: 3 (initialization error).

But there is a error,it extremely confuse me.I have no idea to fix it. Can you give some ideas to me ? Error log as follow:

File "utils/yolo_with_plugins.py", line 237, in _load_engine
    with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:
AttributeError: __enter__

hnsd@hnsd-desktop:~/Projects/tensorrt_demos$ ./app_yolov4
onnx_file is  exist !!
engine_file is  exist !!
Traceback (most recent call last):
  File "app_yolov4.py", line 141, in <module>
    test()
  File "app_yolov4.py", line 20, in test
    trt_yolov4 = load_yolov4_model(detection_id, width)
  File "port_trt_yolov4.py", line 80, in load_yolov4_model
    trt_yolov4 = TrtYOLO(detection_id, (width, width),  class_num, cuda_ctx=None)  # 416 or 608
  File "utils/yolo_with_plugins.py", line 253, in __init__
    self.engine = self._load_engine()
  File "utils/yolo_with_plugins.py", line 237, in _load_engine
    with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:
AttributeError: __enter__
[2219] Failed to execute script app_yolov4
Exception ignored in: <bound method TrtYOLO.__del__ of <utils.yolo_with_plugins.TrtYOLO object at 0x7f60ef8320>>
Traceback (most recent call last):
  File "utils/yolo_with_plugins.py", line 267, in __del__
    del self.outputs
AttributeError: outputs

jkjung-avt, 2020-12-26:
> File "utils/yolo_with_plugins.py", line 237, in _load_engine
> with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:
> AttributeError: \_\_enter\_\_

Referring to https://github.com/jkjung-avt/tensorrt_demos/blob/31e02139ad30987824cf84e3c1b81dbf25eea762/utils/yolo_with_plugins.py#L268

My original code would look for the file, say, "yolo/yolov4-custom-416.trt" and open/deserialize it to a TensorRT engine.  The AttributeError is likely due to _load_engine() failing to find the file.

You could consider to:

* modify the file path or make sure you put the trt (engine) file to the expected path, or
* add the trt file into pyinstaller package ("--add_data").

Alphonses, 2020-12-28:
thanks,I try it now.

tushardhadiwal, 2020-12-28:
> Thank you for your reply. I fix the error by add pycuda into pyinstaller spec:[TensorRT] ERROR: /home/jenkins/workspace/TensorRT/helpers/rel-7.1/L1_Nightly_Internal/build/source/rtSafe/resources.h (460) - Cuda Error in loadKernel: 3 (initialization error).
> 
> But there is a error,it extremely confuse me.I have no idea to fix it. Can you give some ideas to me ? Error log as follow:
> 
> File "utils/yolo_with_plugins.py", line 237, in _load_engine
> with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:
> AttributeError: "**enter**"
> 
> hnsd@hnsd-desktop:~/Projects/tensorrt_demos$ ./app_yolov4
> onnx_file is exist !!
> engine_file is exist !!
> Traceback (most recent call last):
> File "app_yolov4.py", line 141, in
> test()
> File "app_yolov4.py", line 20, in test
> trt_yolov4 = load_yolov4_model(detection_id, width)
> File "port_trt_yolov4.py", line 80, in load_yolov4_model
> trt_yolov4 = TrtYOLO(detection_id, (width, width), class_num, cuda_ctx=None) # 416 or 608
> File "utils/yolo_with_plugins.py", line 253, in **init**
> self.engine = self._load_engine()
> File "utils/yolo_with_plugins.py", line 237, in _load_engine
> with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:
> AttributeError: **enter**
> [2219] Failed to execute script app_yolov4
> Exception ignored in: <bound method TrtYOLO.**del** of <utils.yolo_with_plugins.TrtYOLO object at 0x7f60ef8320>>
> Traceback (most recent call last):
> File "utils/yolo_with_plugins.py", line 267, in **del**
> del self.outputs
> AttributeError: outputs

@Alphonses 
Can you share what exactly did you do to solve the cuda initialization error?
What file did you add to the pyinstaller spec?

jkjung-avt, 2020-12-31:
@tushardhadiwal  I already replied: https://github.com/jkjung-avt/tensorrt_demos/issues/298#issuecomment-751316394

@Alphonses  Any update on this?  If there's no further question, I'd like to close this issue.

tushardhadiwal, 2020-12-31:
> @tushardhadiwal I already replied: [#298 (comment)](https://github.com/jkjung-avt/tensorrt_demos/issues/298#issuecomment-751316394)
> 
> @Alphonses Any update on this? If there's no further question, I'd like to close this issue.

@jkjung-avt I meant, can you throw some clarity on how alphonses resolved the original issue, and what files do I need to add to pyinstaller spec? I tried adding pucuda install location, it still gives same error. I will be very grateful if you can show a pyinstaller command that helps pack this project into a exe.

Alphonses, 2020-12-31:
> > @tushardhadiwal I already replied: [#298 (comment)](https://github.com/jkjung-avt/tensorrt_demos/issues/298#issuecomment-751316394)
> > @Alphonses Any update on this? If there's no further question, I'd like to close this issue.
> 
> @jkjung-avt I meant, can you throw some clarity on how alphonses resolved the original issue, and what files do I need to add to pyinstaller spec to resolve cuda initialization error? I tried adding pucuda install location and bunch of other things, it still gives same error. I will be very grateful if you can show a pyinstaller command that helps pack this project into a exe.

pyinstaller -F --hidden-import=cv2 --hidden-import=onnx.onnx_cpp2py_export --hidden-import=tensorrt  --hidden-import=pycuda  --paths="/usr/local/lib/python3.6/dist-packages/cv2/python-3.6" trt_yolo.py

Just  added  "--hidden-import=pycuda" to   pyinstaller. It works.

tushardhadiwal, 2020-12-31:
@Alphonses The same command that you posted doesn’t work for me. How did you install pycuda?
I used the script provided by @jkjung-avt for installation.

Alphonses, 2020-12-31:
@jkjung-avt     

**- I have already added the trt file into pyinstaller package  with "--add-data".

- I have already modified  the file path and make sure  put the trt (engine) file to the expected path.

But  this error is still exist.** .    It  seems that the trt file has not been added into the generated executed file by pyinstaller.

This question is extremely strange, any  other ideas   thank you sincerely.

hnsd@hnsd-desktop:~/Projects/tensorrt_demos$ ./app_yolov4
onnx_file is exist !!
engine_file is exist !!
Traceback (most recent call last):
File "app_yolov4.py", line 141, in
test()
File "app_yolov4.py", line 20, in test
trt_yolov4 = load_yolov4_model(detection_id, width)
File "port_trt_yolov4.py", line 80, in load_yolov4_model
trt_yolov4 = TrtYOLO(detection_id, (width, width), class_num, cuda_ctx=None) # 416 or 608
File "utils/yolo_with_plugins.py", line 253, in init
self.engine = self._load_engine()
File "utils/yolo_with_plugins.py", line 237, in _load_engine
with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:
AttributeError: enter
[2219] Failed to execute script app_yolov4
Exception ignored in: <bound method TrtYOLO.del of <utils.yolo_with_plugins.TrtYOLO object at 0x7f60ef8320>>
Traceback (most recent call last):
File "utils/yolo_with_plugins.py", line 267, in del
del self.outputs
AttributeError: outputs

Alphonses, 2020-12-31:
> @Alphonses The same command that you posted doesn’t work for me. How did you install pycuda?
> I used the script provided by @jkjung-avt for installation.I

pip3 install pycuda==2019.1.0   for me.

jkjung-avt, 2021-01-01:
> with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:

Try add some debugging code (print statements) to verify whether `open(TRTbin, 'rb')` or instantiation of `trt.Runtime(self.trt_logger)` has failed.

I myself have limited experience with pyinstaller.  Check if [this RealPython tutorial](https://realpython.com/pyinstaller-python/) is helpful.

Alphonses, 2021-01-04:
Hello,@jkjung-avt . I had added some debugging code (print statements) to verify whether open(TRTbin, 'rb') or instantiation of trt.Runtime(self.trt_logger) has failed.

**1. changed  the "yolo_with_plugins.py" code:**

def _load_engine(self):      
        TRTbin = "/home/hnsd/Projects/tensorrt_demos/model_files/10000.trt"
        with open(TRTbin, 'rb') as f:
            fr = f.read()
        print("===================1=======================")
        with trt.Runtime(self.trt_logger) as runtime:
            print("===================2=======================")
            return runtime.deserialize_cuda_engine(fr)
        print("===================3=======================")

**2.I packaged by pyinstaller command:**
pyinstaller -F --add-data="10000.trt:/home/hnsd/Projects/tensorrt_demos/model_files/" --hidden-import=cv2 --hidden-import=onnx.onnx_cpp2py_export --hidden-import=tensorrt   --paths="/usr/local/lib/python3.6/dist-packages/cv2/python-3.6":"/home/hnsd/.local/lib/python3.6/site-packages/pycuda/":"/usr/lib/python3.6/dist-packages/tensorrt/"  app_yolov4.py 

**3. But I run,the error is still  exist ! It is similarly instantiation of trt.Runtime(self.trt_logger) has failed. How can I fix it.**

hnsd@hnsd-desktop:~/Projects/tensorrt_demos$ ./app_yolov4 
onnx_file is  exist !!
engine_file is  exist !!
==========================================TRTbin: /home/hnsd/Projects/tensorrt_demos/model_files/10000.trt
===================1=======================
Traceback (most recent call last):
  File "app_yolov4.py", line 141, in <module>
    test()
  File "app_yolov4.py", line 20, in test
    trt_yolov4 = load_yolov4_model(detection_id, width)
  File "port_trt_yolov4.py", line 80, in load_yolov4_model
    trt_yolov4 = TrtYOLO(detection_id, (width, width),  class_num, cuda_ctx=None)  # 416 or 608
  File "utils/yolo_with_plugins.py", line 262, in __init__
    self.engine = self._load_engine()
  File "utils/yolo_with_plugins.py", line 242, in _load_engine
    with trt.Runtime(self.trt_logger) as runtime:
AttributeError: __enter__

Alphonses, 2021-01-04:
@tushardhadiwal   Hi.This way may help you.

`hnsd@hnsd-desktop:~/Projects/tensorrt_demos$ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pycuda
>>> print(pycuda.__file__)
/home/hnsd/.local/lib/python3.6/site-packages/pycuda/__init__.py`

you can use pyinstaller -F --paths="pycuda-path".
`pyinstaller -F  --paths="/home/hnsd/.local/lib/python3.6/site-packages/pycuda/"`

jkjung-avt, 2021-01-04:
It's apparent that the failure is on this line.  I think the `AssertionError: __enter__` is a result of `trt.Runtime(self.trt_logger)` failing and returning `None`.  So python throws an exception saying that `None` does not have the `__enter__` attribute.

```
        with trt.Runtime(self.trt_logger) as runtime:
```

The reason why `trt.Runtime(self.trt_logger)` fails is, again, likely due to missing modules/libraries.   Please find those missing modules/libraries which TensorRT depends on, and add them into pyinstaller spec.

Alphonses, 2021-01-04:
@jkjung-avt. I think. There is no missing modules/libraries to "trt.Logger" and "trt.Runtime".  so it confuse me for this error.
`self.trt_logger = trt.Logger(trt.Logger.INFO)
self.engine = self._load_engine()`
```
def _load_engine(self):      
        TRTbin = "/home/hnsd/Projects/tensorrt_demos/model_files/10000.trt"
        print("==========================================TRTbin:",TRTbin)
        with open(TRTbin, 'rb') as f:
            fr = f.read()
        print("===================1=======================")
        with trt.Runtime(self.trt_logger) as runtime:
            print("===================2=======================")
            return runtime.deserialize_cuda_engine(fr)
        print("===================3=======================")
```

tushardhadiwal, 2021-01-04:
> The reason why `trt.Runtime(self.trt_logger)` fails is, again, likely due to missing modules/libraries. Please find those missing modules/libraries which TensorRT depends on, and add them into pyinstaller spec.

Any pointers on how to find what libraries or modules are missing? Or the modules which are responsible for this to fail

Alphonses, 2021-01-04:
@tushardhadiwal You can according to errors to fix them.But you need to package the main libraries or modules firstly.

jkjung-avt, 2021-01-04:
> I think. There is no missing modules/libraries to "trt.Logger" and "trt.Runtime".

How do you make sure that?

Try to package the following code to see if it works.

```
import tensorrt as trt
trt_logger = trt.Logger(trt.Logger.INFO)
trt_runtime = trt.Runtime(trt_logger)
assert trt_runtime is not None
```

tushardhadiwal, 2021-01-04:
> How do you make sure that?
>
> Try to package the following code to see if it works.
>
> ```
> import tensorrt as trt
> trt_logger = trt.Logger(trt.Logger.INFO)
> trt_runtime = trt.Runtime(trt_logger)
> assert [...]

@jkjung-avt I packaged this code with simple pyinstaller trial.py command. And ran the exe, it works!

Alphonses, 2021-01-04:
> How do you make sure that?
>
> Try to package the following code to see if it works.
>
> ```
> import tensorrt as trt
> trt_logger = trt.Logger(trt.Logger.INFO)
> trt_runtime = trt.Runtime(trt_logger)
> assert [...]

I have packaged .
`pyinstaller -F test1.py` 
It work for me.

Alphonses, 2021-01-04:
1.I changed the code to the follow  "test1.py". 
```
import tensorrt as trt
trt_logger = trt.Logger(trt.Logger.INFO)
trt_runtime = trt.Runtime(trt_logger)
with trt.Runtime(trt_logger) as runtime:
    print("==========================================")
assert trt_runtime is not None
```
`hnsd@hnsd-desktop:~/Projects/tensorrt_demos/test$  pyinstaller -F test1.py`

2.run. It  also works. Print log is follow:
`hnsd@hnsd-desktop:~/Projects/tensorrt_demos/test/dist$ ./test1 
==========================================`

3. So it is extremely strange.

Alphonses, 2021-01-04:
1.I changed the code to the follow  "test1.py". 
```
import tensorrt as trt
trt_logger = trt.Logger(trt.Logger.INFO)
trt_runtime = trt.Runtime(trt_logger)
with trt.Runtime(trt_logger) as runtime:
    print("==========================================")
assert trt_runtime is not None
```
`hnsd@hnsd-desktop:~/Projects/tensorrt_demos/test$  pyinstaller -F test1.py`

2.run. It  also works. Print log is follow:
`hnsd@hnsd-desktop:~/Projects/tensorrt_demos/test/dist$ ./test1 
==========================================`

3. So it is extreme ly strange.

jkjung-avt, 2021-01-04:
> 3. So it is extremely strange.

Indeed.

Could you set TensorRT logger level to VERBOSE to see if it spits out any meaningful error message?

```
        self.trt_logger = trt.Logger(trt.Logger.VERBOSE)
```

https://github.com/jkjung-avt/tensorrt_demos/blob/8b528c1c344033c6df182770b14d18b023d09f00/utils/yolo_with_plugins.py#L285

Alphonses, 2021-01-04:
**1.I changed the code,the  "AttributeError: __enter__"  is gone.Is it must use "with" ?**
`with trt.Runtime(self.trt_logger) as runtime:`   to  `runtime = trt.Runtime(self.trt_logger)`

**2.But I meet another error:**
 ` ` `
[TensorRT] ERROR: /home/jenkins/workspace/TensorRT/helpers/rel-7.1/L1_Nightly_Internal/build/source/rtSafe/resources.h (460) - Cuda Error in loadKernel: 3 (initialization error)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
```
**3. In order to make sure reason of the error . I changed the code as follow:**
 ` ` `
class TrtYOLO(object):
    """TrtYOLO class encapsulates things needed to run TRT YOLO."""

    def _load_engine(self):
        TRTbin = "/home/hnsd/Projects/tensorrt_demos/model_files/10000.trt"
        print("==========================================TRTbin:",TRTbin)
        with open(TRTbin, 'rb') as f:
            fr = f.read()
        assert fr is not None
        print("===================1=======================")
        runtime = trt.Runtime(self.trt_logger)
        print("===================2=======================")
        return runtime.deserialize_cuda_engine(fr)
    
        #with open(TRTbin, 'rb') as f, trt.Runtime(self.trt_logger) as runtime:
        #    return runtime.deserialize_cuda_engine(f.read())

    def __init__(self, model, input_shape, category_num, cuda_ctx=None):
        """Initialize TensorRT plugins, engine and conetxt."""
        self.model = model
        self.input_shape = input_shape
        self.category_num = category_num
        self.cuda_ctx = cuda_ctx
        if self.cuda_ctx:
            self.cuda_ctx.push()

        self.inference_fn = do_inference_v2         
        self.trt_logger = trt.Logger(trt.Logger.VERBOSE)   # INFO
        self.engine = self._load_engine()
        print("===================3=======================")
        assert self.engine is not None
        try:
            self.context = self.engine.create_execution_context()
            print("===================4=======================")
            self.inputs, self.outputs, self.bindings, self.stream = \
                allocate_buffers(self.engine)
        except Exception as e:
            raise RuntimeError('fail to allocate CUDA resources') from e
        finally:
            if self.cuda_ctx:
                self.cuda_ctx.pop()

```

**Then run it. The error log as follow:**
 ` ` `
hnsd@hnsd-desktop:~/Projects/tensorrt_demos$ ./app_yolov4 
onnx_file is  exist !!
engine_file is  exist !!
==========================================TRTbin: /home/hnsd/Projects/tensorrt_demos/model_files/10000.trt
===================1=======================
===================2=======================
[TensorRT] ERROR: /home/jenkins/workspace/TensorRT/helpers/rel-7.1/L1_Nightly_Internal/build/source/rtSafe/resources.h (460) - Cuda Error in loadKernel: 3 (initialization error)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
===================3=======================
Traceback (most recent call last):
  File "app_yolov4.py", line 141, in <module>
    test()
  File "app_yolov4.py", line 20, in test
    trt_yolov4 = load_yolov4_model(detection_id, width)
  File "port_trt_yolov4.py", line 80, in load_yolov4_model
    trt_yolov4 = TrtYOLO(detection_id, (width, width),  class_num, cuda_ctx=None)  # 416 or 608
  File "utils/yolo_with_plugins.py", line 268, in __init__
    assert self.engine is not None
AssertionError
```
**4. It can be seen from the above log. The main error reason is the code command :**
(1)`assert fr is not None`  has no "AssertionError".

(2)but   ` runtime.deserialize_cuda_engine(fr)`  appeared the "Deserialize the cuda engine failed".

(3) `assert self.engine is not None`  appeared the "AssertionError".

**so, could trt model  not read into and to Deserialize the cuda engine?**

jkjung-avt, 2021-01-04:
> (2) but runtime.deserialize_cuda_engine(fr) appeared the "Deserialize the cuda engine failed".

The trt file (serialized TensorRT engine) should be a **binary** file.  Check if pyinstaller has somehow treated it as a text file.  I think you could add some command-line flags to let pyinstaller know that's a binary file.

Alphonses, 2021-01-05:
**Hello, @jkjung-avt .  I have tried pyinstaller by `--add-binary="yolov4.trt:model_files", but it can not work` .I found this problem after my morning investigation with google.The main error as follow:**
**TensorRT + PyInstaller on Nvidia Jetson. Deserialize engine causes Cuda Error in loadKernel: 3 (initialization error)**
`[TensorRT] ERROR: /home/jenkins/workspace/TensorRT/helpers/rel-7.1/L1_Nightly_Internal/build/source/rtSafe/resources.h (460) - Cuda Error in loadKernel: 3 (initialization error)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.`

**1.Many learners meet this problem.Can you  fix it on your repo with your abundant experience or tell me how can I fix it on your repo.  Sincere thanks for your patience!**
[https://stackoverflow.com/questions/64608796/tensorrt-pyinstaller-on-nvidia-jetson-nano-deserialize-engine-causes-cuda-err](url)
[https://github.com/pyinstaller/pyinstaller/issues/5191](url)

jkjung-avt, 2021-01-08:
@Alphonses  I did a simple test.  Deserializing a TensorRT engine inside pyinstaller works OK.

1. Create a directory for testing.  Copy over the TensorRT engine file and the plugin.
    ```
    $ mkdir pyinstaller_test
    $ cd pyinstaller_test
    $ cp ${HOME}/project/tensorrt_demos/yolo/yolov4-608.trt .
    $ cp ${HOME}/project/tensorrt_demos/plugins/libyolo_layer.so .
    ```

2. Create a "test.py" with the following code.
    ```
    """test.py
    """
    
    import ctypes
    import tensorrt as trt
    
    ctypes.cdll.LoadLibrary('libyolo_layer.so')
    
    trt_logger = trt.Logger(trt.Logger.INFO)
    with open('yolov4-608.trt', 'rb') as f, trt.Runtime(trt_logger) as runtime:
        engine = runtime.deserialize_cuda_engine(f.read())
    print('Done.')
    ```

3. Build the code with pyinstaller and test it.
    ```
    $ pyinstaller -F test.py
    $ ./dist/test
    ```

If things work correctly, you'd see "Done." at the end.

jkjung-avt, 2021-01-08:
@Alphonses  Following up on the previous post:

1. I don't need to add the TensorRT engine file and the libyolo_layer.so into the pyinstaller executable.  They could be packaged as separate files.  (I think it's also OK to include them in the pyinstaller executable.)

2. I suspect the reason why you encounter error while deserializing the engine is because libyolo_layer.so is missing.  Please check that.

> [TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.

Alphonses, 2021-01-11:
Hello,@jkjung-avt. I found the problem. It  works well. Sincere thanks for your patience!

tushardhadiwal, 2021-01-11:
@Alphonses

What exactly worked for you?
can you share the full pyinstaller command?

Alphonses, 2021-01-12:
Hello,@jkjung-avt @tushardhadiwal   

Delete   "import pycuda.autoinit" in trt_yolo.py.

Then  add  "import pycuda.autoinit"  to  utils/yolo_with_plugins.py.  

It will work well.

You can close the issue . @jkjung-avt  ,thank you.

jkjung-avt, 2021-01-12:
> Delete "import pycuda.autoinit" in trt_yolo.py.
> Then add "import pycuda.autoinit" to utils/yolo_with_plugins.py.
> It will work well.

@Alphonses  Thanks for sharing the solution :-)

closed 2021-01-12 by jkjung-avt