jkjung-avt/tensorrt_demos #299: demo4: onnx_to_tensorrt.py failed

Robotzeroo, 2020-12-19:
@jkjung ,thank you very much for your efforts in this Repo.
My environment is tensorrt 6.0.1.10, and when I was running "Sudo Python3 onnx_to_tensorrt.py-v-m yolov3-416", I made the following mistake. This let me confused. Can you give me some ideas? Thank a lot!!

error log is :
Loading the ONNX file...
Adding yolo_layer plugins...
Building an engine.  This would take a while...
(Use "--verbose" or "-v" to enable verbose logging.)
python3: yolo_layer.cu:116: virtual nvinfer1::Dims nvinfer1::YoloLayerPlugin::getOutputDimensions(int, const nvinfer1::Dims*, int): Assertion `inputs[0].d[1] == mYoloHeight' failed.
Aborted

jkjung-avt, 2020-12-20:
Are you using the pre-trained (downloaded) yolov3 model?  If so, could you check and make sure you have `width=416` and `height=416` set correctly n the "yolov3-416.cfg" file?

Robotzeroo, 2020-12-20:
> Are you using the pre-trained (downloaded) yolov3 model? If so, could you check and make sure you have `width=416` and `height=416` set correctly n the "yolov3-416.cfg" file?

closed 2020-12-20 by Robotzeroo

Robotzeroo, 2020-12-20:
It is actually because the width and height in yolov3-416.cfg are not equal to 416. It seems that I haven't finished executing ./download_yolo.sh. After i corrected this error, it has successfully run on tx2 now. Thanks a lot!!