jkjung-avt/tensorrt_demos #300: builder.build_engine - Driver error with onnx_to_tensorrt.py

ROBYER1, 2020-12-21:
When running `python3 yolo_to_onnx.py -m yolov4-416` from https://github.com/jkjung-avt/tensorrt_demos#demo-5-yolov4

I get these errors around the builder.build_engine lines:

> Traceback (most recent call last):
>   File "onnx_to_tensorrt.py", line 195, in <module>
>     main()
>   File "onnx_to_tensorrt.py", line 183, in main
>     engine = build_engine(
>   File "onnx_to_tensorrt.py", line 154, in build_engine
>     engine = builder.build_engine(network, config)
> RuntimeError: Driver error: 

Linux Ubuntu 20.04 LTS on Windows 10 WSL 2
Cuda 11-1
Cuda compilation tools, release 11.1, V11.1.105 Build cuda_11.1.TC455_06.29190527_0
Cudnn 8.0.5.39
Tensorrt 7.2.2.3
Python 3.8 system install default

Mentioned here too as I saw the same error there :https://github.com/NVIDIA/TensorRT/issues/875#issuecomment-749066732

jkjung-avt, 2020-12-22:
I'm not sure whether CUDA and TensorRT would work correctly in a Ubuntu on WSL2 environment.  Could you try out a couple of the [official TensorRT samples](https://docs.nvidia.com/deeplearning/tensorrt/sample-support-guide/index.html) on your system first?

jkjung-avt, 2020-12-24:
I add some references here:

* [CUDA on WSL User Guide](https://docs.nvidia.com/cuda/wsl-user-guide/index.html)
* [CUDA on Windows Subsystem for Linux (WSL) - Public Preview](https://developer.nvidia.com/cuda/wsl)

As of today, the CUDA driver/toolkit on WSL is still a Public Preview release.  So I won't be surprised to see if there are some problems running it on certain platforms...

ROBYER1, 2020-12-24:
I'll close this issue for now as I don't fully need Tensorrt, if I need it I will install it on my windows main partition (if I can)

closed 2020-12-24 by ROBYER1