jkjung-avt/tensorrt_demos #303: yolov4_tiny-608X608 - Yolo to ONXX conversion error.

parameswar-ds, 2020-12-23:
Parsing DarkNet cfg file...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 955, in <module>
    main()
  File "yolo_to_onnx.py", line 896, in main
    layer_configs = parser.parse_cfg_file(cfg_file_path)
  File "yolo_to_onnx.py", line 94, in parse_cfg_file
    layer_dict, layer_name, remainder = self._next_layer(remainder)
  File "yolo_to_onnx.py", line 150, in _next_layer
    param_type, param_value = self._parse_params(param_line)
  File "yolo_to_onnx.py", line 177, in _parse_params
    param_value = float(param_value_raw)
ValueError: could not convert string to float: '1.5y'

jkjung-avt, 2020-12-23:
This is because the parser code in "yolo_to_onnx.py" failed to handle something in your cfg file.  You could check where in your cfg file the "1.5y" pattern could be found.  Fix it.  Then the script should work.

parameswar-ds, 2020-12-23:
@jkjung-avt 
Thanks man,
but, Getting large detected area instead of exact ROI, 
I have trained a custom tiny YOLOv4 with 416X416 reslolution, but Running inference on 1080X1080 resolution.
Will that going to effect?

jkjung-avt, 2020-12-23:
> I have trained a custom tiny YOLOv4 with 416X416 resolution, but Running inference on 1080X1080 resolution.

That won't work too well.  It's better to train and inference with the same resolution.

closed 2020-12-25 by jkjung-avt