jkjung-avt/tensorrt_demos #304: Yolov4-csp support

tushardhadiwal, 2020-12-23:
Hello,

Thank you for your great work on making tensorrt implementations of yolo models.
I want to know if these scripts can be used to convert Yolov4-csp models to tensorrt.

Thanks

jkjung-avt, 2020-12-23:
Just did.  I merged my "scaled_yolov4" branch into master, which now supports "yolov4-csp" and "yolov4x-mish" models.

Note that, as stated in the README, you'll have to re-build the "yolo_layer" plugin after you pull the latest code.

tushardhadiwal, 2020-12-23:
@jkjung-avt Thanks for updating the repo.

I tried yolov4-csp-512 on jetson nano. 
I recompiled plugin and built trt engine.
I am not able to get FPS more than 1.80 with trt_yolo.py on jetson nano with MaxN mode.

How can i reproduce this 4.26FPS?

yolov4-csp-512 (FP16) | 0.436 | 0.630 | 4.26

jkjung-avt, 2020-12-23:
1. Be sure to do `sudo jetson_clocks` as well.
2. I use [dog.jpg](https://github.com/AlexeyAB/darknet/blob/master/data/dog.jpg) to measure FPS.
3. For "yolov4-csp" and "yolov4x-mish" models, run "trt_yolo.py" with the "--letter_box" (or "-l") option.

tushardhadiwal, 2020-12-23:
@jkjung-avt I have jetson_clocks running and I used --letter_box option.
I am using a video that I used with other models earlier. For all yolov4 and yolov4tiny models I got similar FPS to what shown in ReadMe. But in case of CSP models its far lower.

jkjung-avt, 2020-12-23:
> But in case of CSP models its far lower.

Thanks for letting me know.  I will double check when I have a chance.

jkjung-avt, 2020-12-24:
I tested TensorRT "yolov4-csp-512" engine on my Jetson Nano DevKit (JetPack-4.4, TensorRT 7.1) again.  I'm able to get 4.24~4.26 FPS.

closed 2020-12-25 by jkjung-avt

Tetsujinfr, 2021-03-18:
Hi
Why the MAP score are worse for yolov4-csp 512 than for yolov4-412?
Per AlexAB blog [here](https://alexeyab84.medium.com/scaled-yolo-v4-is-the-best-neural-network-for-object-detection-on-ms-coco-dataset-39dfa22fa982), scaled yolov4 is supposed to achieve better accuracy at equivalent latency no?

jkjung-avt, 2021-03-18:
@Tetsujinfr  I did my best to make sure my TensorRT implementation of yolov4-csp is correct.  If you find issues in my code, do let me know.

Otherwise, I have no idea why the mAP is lower than expected.  I did not verify mAP of the original yolov4-csp-512.cfg and yolov4-csp-512.weights myself.

Tetsujinfr, 2021-03-18:
Ok np. Unfortunately Ido not have the skills to check your excellent code, just I thought csp would petform better after reading AlexAB article. Either tensorrt is adversely degrading the scaled model or there is something to improve on your current implementation. I was just checking if csp models here were indeed corresponding to scaled yolo4.
Thanks for your great repo and excellent work as always.