jkjung-avt/tensorrt_demos #306: Yolov4 with batch greater than 1

tungth07, 2020-12-24:
Hi @jkjung-avt 

First of all, thanks you for your contributions. Currently I am using your project to convert yolov4-tiny to tensorrt.

If the MAX_BATCH_SIZE=1 everything is OK. But when I set MAX_BATCH_SIZE > 1, and modified inputs and outputs as you recommend in some previous issues, but the result is not OK.

I only received result of first image in batch, and the others is zero. 

Pls give me some advice to resolved this problem.

Thanks & Best Regards

jkjung-avt, 2020-12-24:
I made necessary modifications to the "yolo_layer" plugin in the [batched_yolo](https://github.com/jkjung-avt/tensorrt_demos/tree/batched_yolo) branch.  You'll still have to modify the pre-processing and post-processing code to make it work.

I cannot offer more help than this.

tungth07, 2020-12-25:
Thanks for your help. I have solved my problem.

Thanks & Merry Chrismast

closed 2020-12-25 by tungth07

papalotis, 2020-12-26:
hey @tungth07 can you give any hints as to how you changed the pre-,postprocessing sections

tungth07, 2020-12-26:
with pre-processing, you can do as get_batch() function in calibrator.py with batch size greater than 1.

In post process you do a loop according to the number of input images

That's all

jkjung-avt, 2021-01-01:
For the record, I have updated the plugin code in the master branch to support batch_size greater than 1 as well.  I would suggest people to use the code in the master branch.

https://github.com/jkjung-avt/tensorrt_demos/commit/793d7ae9c74b51007296bb63cdd1bc6bd0e04e8e

papalotis, 2021-01-14:
I've been trying to figure out the postprocessing steps but for some reason I cannot figure it out. @tungth07 can you post a snippet over how you are looping over the trt_outputs? Depending on the batch the two output tensors have a different length but I cannot figure out how I should split them

liermvl, 2021-03-02:
@tungth07, @papalotis, @jkjung-avt  would it be possible to show how to copy a set of batch images to the device?
I followed the code in get_batch() function in calibrator.py, where I prepare a batch of images as follows:

```
# Prepare a batch
assert len(imgs) == self.batch_size
resized_batch = []
for img in imgs:
    resized_batch.append(_preprocess_yolo(img, self.input_shape, letter_box))
resized_batch = np.stack(resized_batch)
assert resized_batch.nbytes == self.blob_size
```

Now I am not sure how to copy the batch of images to the host. get_batch() function in calibrator.py is copying the batch at ones, so in my case that would be as follows:
`self.inputs[0].host = np.ascontiguousarray(img_resized)`

But do_inference_v2() in yolo_with_plugins.py has the option to take multiple inputs, and copy them one for one:
```
    """
    This function is generalized for multiple inputs/outputs for full
    dimension networks.
    Inputs and outputs are expected to be lists of HostDeviceMem objects.
    """
    # Transfer input data to the GPU.
    [cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
```

So I could do something as this then:
```
for i in range(self.batch_size):
     self.inputs[i].host = np.ascontiguousarray(resized_batch[i])
```

@jkjung-avt Any suggestions are welcome, I am also trying to understand the code and interface the cuda device.

jkjung-avt, 2021-03-03:
> But do_inference_v2() in yolo_with_plugins.py has the option to take multiple inputs, and copy them one for one:

I think you misunderstood the code.  The yolo model only takes 1 input: preprocessed image (either single or batched).  So there would always be only 1 item (tensor) in the "inputs" list.

In contrast, there would be 2 or 3 items in the "outputs" (yolo layers), depending whether you are running the "tiny" model or not.

> show how to copy a set of batch images to the device?

1. Make sure you set max_batch_size when you build the TensorRT engine.
2. [allocate_buffers()](https://github.com/jkjung-avt/tensorrt_demos/blob/94d06b356822dd230a969d8697dc8c3e55f0dfd6/utils/yolo_with_plugins.py#L203) would then allocate the correct sizes of input/output GPU buffers for you.
3. In [TrtYOLO.detect()](https://github.com/jkjung-avt/tensorrt_demos/blob/94d06b356822dd230a969d8697dc8c3e55f0dfd6/utils/yolo_with_plugins.py#L311), stack a batch of preprocessed images into 1 contiguous numpy array, shaped as (batch_size, 3, input_height, input_width).
4. [do_inference_v2()](https://github.com/jkjung-avt/tensorrt_demos/blob/94d06b356822dd230a969d8697dc8c3e55f0dfd6/utils/yolo_with_plugins.py#L311) would do the right thing for you.  No modification of code is needed here.
5. Post-process outputs of the TensorRT engine, one image at a time.  Note that the 2 or 3 outputs would also contain batched outputs for all input images.

liermvl, 2021-03-09:
@jkjung-avt I was able to make work steps 1-4. But step 5 I am not sure, since, for every batch size step, B=2, B=4, the output is quadrupled, for yolov4 416x416 model, in the master branch. But this is strange since I would expect that it only doubles, could this be a bug somewhere in the conversion of the plugin? See the numbers below for an example:

E.g. when looking at the output for batch size = 1, then len() of the tensors is: (56784, 14196, 3549). But for B=2 this is: (227136, 56784, 14196) and B=4: (908544, 227136, 56784). 

What would be the best way to extract the tensor out of the batched tensor? E.g. `trt_outputs[0][int(len(trt_outputs[0]/2)):len(trt_outputs[0]]
`

liermvl, 2021-03-09:
@jkjung-avt I was able to make work steps 1-4. But step 5 I am not sure, since, for every batch size step, B=2, B=4, the output is quadrupled, for yolov4 416x416 model, in the maste rbranch.
E.g. when looking at the output for batch size = 1, then len( of the tensors is: (56784, 14196, 3549). But for B=2 this is: (227136, 56784, 14196) and B=4: (908544, 227136, 56784).

What would be the best way to extract the tensor out of the batched tensor? E.g. `trt_outputs[0][int(len(trt_outputs[0]/2)):len(trt_outputs[0]]
`

jkjung-avt, 2021-03-09:
> But step 5 I am not sure, since, for every batch size step, B=2, B=4, the output is quadrupled, for yolov4 416x416 model, in the master branch.

The output tensor sizes should have been doubled when you double input batch size.

For example, you get 3 output tensors for B=1 of shapes: (1, 56784, 1, 1), (1, 14196, 1, 1), and (1, 3549, 1, 1).

Then you should get 3 output tensors for B=2 of shapes: (2, 56784, 1, 1), (2, 14196, 1, 1), and (2, 3549, 1, 1).

Please double check: (1) you have built the TensorRT engine with the correct "max_batch_size"; (2) you have fed the right size of input tensor to the TensorRT engine at inference time.  One additional thing you could check is whether the input/output "binding volumes" (tensor sizes) are as expected.

https://github.com/jkjung-avt/tensorrt_demos/blob/53684d733693e68f66b5c9987dd6c293aac76b49/utils/yolo_with_plugins.py#L212-L213

> What would be the best way to extract the tensor out of the batched tensor?

For example,

```
        batched_trt_outputs = self.inference_fn(......)
        batched_trt_outputs = [o.reshape((batch_size, -1)) for o in batched_trt_outputs]
        for i in range(batch_size):
            trt_outputs = [o[i] for o in batched_trt_outputs]]
            # call _postprocess_yolo() for a single image in the batch
```

jkjung-avt, 2021-03-09:
Hmm...  Could you try this?

Replace:

```
    size = trt.volume(engine.get_binding_shape(binding)) * engine.max_batch_size
```

with this:

```
    size = trt.volume(engine.get_binding_shape(binding))
```

Then check whether the TensorRT engine produce the correct output?

liermvl, 2021-03-10:
Yes, this did the trick indeed, so I guess it is a bug.

My JetPack version == 4.5.1

jkjung-avt, 2021-03-10:
> Yes, this did the trick indeed, so I guess it is a bug.

Thanks for confirming this.  I will find time to fix and test the code.

liermvl, 2021-03-10:
Thank you for your dedication and quick response.

jkjung-avt, 2021-03-12:
I have implemented the fix: https://github.com/jkjung-avt/tensorrt_demos/commit/e95712da4f9f880908546b9b609e52ecd9e870d6

jkjung-avt, 2021-10-12:
@ttducqt  The following 2 commits add a "Concat" layer into the TensorRT YOLO engine.  The "Concat" would concatenate all "yolo" layer outputs and generate only 1 output.  That does not work too well for the batched input cases.

* https://github.com/jkjung-avt/tensorrt_demos/commit/d9a14928508e07dea9a174acc8343075f450ebce#diff-5b3a456b86a643d8b45ac7b6ec42b74626bc6fff14f9b782c51aa92a22377627
* https://github.com/jkjung-avt/tensorrt_demos/commit/1f8773b3f6091d2e3441c0d6ce5714e332421f8d#diff-eb7fe215d31619082d12363996ce41894f01dc0af8065e3dcf7389880b1e05e9

You might want to revert those changes if you'd like to use the code for batched inference.

ttducqt, 2021-10-15:
I made it. Thank you