jkjung-avt/tensorrt_demos #307: Yolov4-tiny on darknet results

cbstryker, 2020-12-24:
This is more of a multi-question issue.

1. I used [these instructions](https://forums.developer.nvidia.com/t/deepstream-sdk-faq/80236/8) from Nvidia's forums, and I can get the process to work. But the detection results compared to pure darknet are fairly terrible. Specifically the "person" class is almost never detected with the TensorRT/Darknet version, whereas the darknet version is nearly infallible. 
2. In your implementation, are you using "libyolo_layer.so" in place of the Nvidia's "libnvdsinfer_custom_impl_Yolo.so" version?

If I use the default "NvDsInferParseCustomYoloV3Tiny" function in Nvidia's version, the detection boxes are tiny and all over the place. If I used the the v4 version from the provided instructions on Nvidia's forums, then I get:

`nvdsparsebbox_Yolo.cpp:342: bool NvDsInferParseYoloV4(const std::vector<NvDsInferLayerInfo>&, const NvDsInferNetworkInfo&, const NvDsInferParseDetectionParams&, std::vector<NvDsInferObjectDetectionInfo>&, const std::vector<float>&, const std::vector<std::vector<int> >&, uint): Assertion scores.inferDims.numDims == 2 failed`

So I'm finding it difficult with how to implement Yolov4-tiny in a way that keeps the same (or at least similar) detection accuracy as native darknet.

Thanks for your time.

jkjung-avt, 2020-12-25:
> 2. In your implementation, are you using "libyolo_layer.so" in place of the Nvidia's "libnvdsinfer_custom_impl_Yolo.so" version?

Basically yes.  The "libnvdsinfer_custom_impl_Yolo.so" includes yolo_layer plugin plus some postprocessing code.  My "libyolo_layer.so" only does yolo_layer calculations (using a CUDA kernel).  I do postprocessing in python.

> If I used the the v4 version from the provided instructions on Nvidia's forums, then I get: [Assertion failed...]

My code in this repo works for yolov4-tiny models.  I have verified mAP of my custom trained model.  The difference between the darknet yolov4-tiny model and the corresponding TensorRT (FP16) engine is within 1~2%.  Feel free to give it a try.

jkjung-avt, 2021-01-04:
Closing this issue since there is no further question.

@cbstryker  In case you'd like to re-open this issue, please convert your yolov4-tiny model by following instructions in this repo and share the results.

closed 2021-01-04 by jkjung-avt