jkjung-avt/tensorrt_demos #309: Issue when converting yolov4 to trt

Lynchez, 2020-12-25:
/content/tensorrt_demos
python3 setup.py build_ext -if
Compiling pytrt.pyx because it changed.
[1/1] Cythonizing pytrt.pyx
running build_ext
building 'pytrt' extension
creating build
creating build/temp.linux-x86_64-3.6
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I/usr/include/python3.6m -c pytrt.cpp -o build/temp.linux-x86_64-3.6/pytrt.o -O3 -std=c++11 -I/usr/local/lib/python3.6/dist-packages/numpy/core/include -I/usr/local/cuda/include -I/usr/local/TensorRT-7.1.3.4/include -I/usr/local/include
In file included from trtNet.cpp:3:0,
                 from pytrt.cpp:670:
trtNet.h:12:10: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
          ^~~~~~~~~~~
compilation terminated.
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
Makefile:4: recipe for target 'all' failed
make: *** [all] Error 1

jkjung-avt, 2020-12-26:
Please modify TensorRT include and lib paths in "setup.py" to point to where your TensortRT headers/libs are installed

https://github.com/jkjung-avt/tensorrt_demos/blob/31e02139ad30987824cf84e3c1b81dbf25eea762/setup.py#L30

https://github.com/jkjung-avt/tensorrt_demos/blob/31e02139ad30987824cf84e3c1b81dbf25eea762/setup.py#L10

Lynchez, 2020-12-26:
Thanks a lot.

closed 2020-12-26 by Lynchez