jkjung-avt/tensorrt_demos #312: what do i need to modify when i have changed the number of anchors in each yolo layer?

IbrahimBond, 2020-12-29:
i am getting this error: 

```
Traceback (most recent call last):
  File "/home/kioskadmin1/Desktop/NLPR/DEV_IFMS_multiprocessing/detection/detection_utils/yolov3.py", line 249, in __init__
    allocate_buffers(self.engine, grid_sizes)
  File "/home/kioskadmin1/Desktop/NLPR/DEV_IFMS_multiprocessing/detection/detection_utils/yolov3.py", line 164, in allocate_buffers
    assert size == grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size
AssertionError
```
cfg file:
```

[net]
# Testing
#batch=1
#subdivisions=1
# Training
batch=64
subdivisions=32
width=288
height=512
channels=3
momentum=0.949
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.001
burn_in=1000
max_batches = 500500
policy=steps
steps=400000,450000
scales=.1,.1

#cutmix=1
mosaic=1

#:104x104 54:52x52 85:26x26 104:13x13 for 416

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=mish

# Downsample

[convolutional]
batch_normalize=1
filters=64
size=3
stride=2
pad=1
activation=mish

[convolutional]
batch_normalize=1
filters=64
size=1
stride=1
pad=1
activation=mish

[route]
layers = -2
[... 1067 lines cut ...]
size=1
stride=1
pad=1
filters=36
activation=linear

[yolo]
mask = 5,6,7,8
anchors = 52, 34,  63, 53,  79, 75, 102,102,  85,190, 131,140, 158,187, 190,264, 222,399
classes=4
num=9
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1
scale_x_y = 1.05
iou_thresh=0.213
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
nms_kind=greedynms
beta_nms=0.6
max_delta=5

```

jkjung-avt, 2020-12-29:
I think this is very likely because you did not specify the correct layers as outputs for your custom model.  Please refer to the code snippet below.

https://github.com/jkjung-avt/tensorrt_demos/blob/2af91c3e6987d9f5e94124a8331bcb4d5f243a23/yolo/yolo_to_onnx.py#L914-L925

IbrahimBond, 2020-12-29:
so the output is wrong because i have changed the masks order in the yolo layers?

can you give an example of how to specify the correct output? 

and how does this affect this snippet of code from the yolo_with_plugins script:

`def get_yolo_grid_sizes(model_name, h, w):
    """Get grid sizes (w*h) for all yolo layers in the model."""
    if 'yolov3' in model_name:
        if 'tiny' in model_name:
            return [(h // 32) * (w // 32), (h // 16) * (w // 16)]
        else:
            return [(h // 32) * (w // 32), (h // 16) * (w // 16), (h // 8) * (w // 8)]
    elif 'yolov4' in model_name:
        if 'tiny' in model_name:
            return [(h // 32) * (w // 32), (h // 16) * (w // 16)]
        else:
            return [(h // 8) * (w // 8), (h // 16) * (w // 16), (h // 32) * (w // 32)]
    else:
        raise ValueError('ERROR: unknown model (%s)!' % args.model)`

jkjung-avt, 2020-12-29:
> for reference my model is yolov4.cfg

Your original description of the problem is not clear.  The AssertionError trace-back shows the python source code as "yolov3.py", which is misleading.

```
Traceback (most recent call last):
  File "/home/kioskadmin1/Desktop/NLPR/DEV_IFMS_multiprocessing/detection/detection_utils/yolov3.py", line 249, in __init__
    allocate_buffers(self.engine, grid_sizes)
  File "/home/kioskadmin1/Desktop/NLPR/DEV_IFMS_multiprocessing/detection/detection_utils/yolov3.py", line 164, in allocate_buffers
    assert size == grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size
AssertionError
```

So, back to trying to resolving the problem for you...

First, make sure you name your cfg/weights files as "yolov4-XXX-288x512" so that "yolo_to_onnx.py" and "onnx_to_tensorrt.py" could handle it correctly.  Reference: [TensorRT YOLOv3 For Custom Trained Models](https://jkjung-avt.github.io/trt-yolov3-custom/)

If that still does not work.  Please print out the values which causes the AssertionError: `size` and `grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size`.  I need that information to guess what the problem might be.

IbrahimBond, 2020-12-29:
you are correct i should put more context. the yolov3 file showing in the traceback is updated to contain the code from your repo that works with yolov4, but i did not change the file name yet. i referred the snippet exactly from your repo so you can better understand what i referring to. 

i have trained the same exact model, but with the default yolov4 anchors and without changing the masks order and it ran fine without the assertion error. the names for .cfg and .weights files were as follows: "yolov4-288x512" which corresponds to the model size and the. trt file was generated without a problem. 

the problem appeared when i trained the model on custom anchors with changing the masks order as shows in the cfg file above. the .trt was generated fine, but the assertion error started appearing upon loading the model.

now for the values:
size: 32256
grid_sizes: [2304, 576, 144]
size == grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size: False

i hope this i clear.

jkjung-avt, 2020-12-29:
I see.  Since you are using yolo layers with 2 and 4 (not 3) anchors:

```
[yolo]
mask = 0,1
anchors = 52, 34,  63, 53,  79, 75, 102,102,  85,190, 131,140, 158,187, 190,264, 222,399
......
[yolo]
mask = 5,6,7,8
anchors = 52, 34,  63, 53,  79, 75, 102,102,  85,190, 131,140, 158,187, 190,264, 222,399
......
```

The assertion is no longer valid...

```
assert size == grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size
```

It should be like the following instead.

```
assert size == grid_sizes[output_idx] * 2 * 7 * engine.max_batch_size or
       size == grid_sizes[output_idx] * 3 * 7 * engine.max_batch_size or
       size == grid_sizes[output_idx] * 4 * 7 * engine.max_batch_size
```

I should find time to fix the code for such cases later on.  For now, I think it's safe for you to remove the `assert` statement.

jkjung-avt, 2020-12-30:
I just double checked.  The latest code in this repo does not have this problem.  The `assert` statement has already been simplified as below.

https://github.com/jkjung-avt/tensorrt_demos/blob/793d7ae9c74b51007296bb63cdd1bc6bd0e04e8e/utils/yolo_with_plugins.py#L203

IbrahimBond, 2020-12-30:
it works perfectly fine now. i will find time to update my code to the latest version in this repo.

thanks for your time.

closed 2020-12-30 by IbrahimBond