jkjung-avt/tensorrt_demos #313: mtcnn problem,  after recognize picture includes one people, it can't recognize picture includes multi person correctly

wavesCHJ, 2020-12-30:
hi @jkjung-avt ，thanks your amazing work and it helps me a lot! 
But i find a problem when running the trt_mtcnn.py . I add some code as below and run it:
`
def main2():
    mtcnn = TrtMtcnn()
    img1 = cv2.imread('curry.jpg')
    dets, landmarks = mtcnn.detect(img1, minsize=40)
    print('len(): {}'.format(len(dets)))
    # img = show_faces(img1, dets, landmarks)
    # cv2.imwrite('pic1.jpg', img)

    img2 = cv2.imread('Avengers.jpg')
    dets, landmarks = mtcnn.detect(img2, minsize=40)
    print('len(): {}'.format(len(dets)))
    # img = show_faces(img2, dets, landmarks)
    # cv2.imwrite('pic2.jpg', img)

if __name__ == '__main__':
    # main()
    main2()
`
the result is:
`
(tensorrt_demos) chenhj@guo-Z10PE:~/tensorrt_demos$ python trt_mtcnn.py 
Deserialize required 1661885 microseconds.
Current optimization profile is: 0. Please ensure there are no enqueued operations pending in this context prior to switching profiles
Deserialize required 3683 microseconds.
Current optimization profile is: 0. Please ensure there are no enqueued operations pending in this context prior to switching profiles
Deserialize required 6849 microseconds.
Current optimization profile is: 0. Please ensure there are no enqueued operations pending in this context prior to switching profiles
len(): 1
len(): 7
`
the number of the face in 'Avengers.jpg'  is incorrect.
But if I only run 'Avengers.jpg' as the follow code, the result is right.
`
def main2():
    mtcnn = TrtMtcnn()
    # img1 = cv2.imread('curry.jpg')
    # dets, landmarks = mtcnn.detect(img1, minsize=40)
    # print('len(): {}'.format(len(dets)))
    # img = show_faces(img1, dets, landmarks)
    # cv2.imwrite('pic1.jpg', img)

    img2 = cv2.imread('Avengers.jpg')
    dets, landmarks = mtcnn.detect(img2, minsize=40)
    print('len(): {}'.format(len(dets)))
    # img = show_faces(img2, dets, landmarks)
    # cv2.imwrite('pic2.jpg', img)

if __name__ == '__main__':
    # main()
    main2()
`
Why this happened?

the curry.jpg is below:
[image]

jkjung-avt, 2020-12-31:
Thanks for reporting this.  I think this refers to the same issue as https://github.com/jkjung-avt/tensorrt_demos/issues/297.

I need to find time to debug this...

jkjung-avt, 2021-02-17:
I have done some more testing of this issue.  Here's what I have found so far.

1. The current trt_mtcnn code works with TensorRT 3 and 4 correctly.  I don't see the reported issue on those older versions of TensorRT.
2. I could reproduce the issue (trt_mtcnn failing to detect all faces after the 1st batch) with TensorRT 7.
3. Following up on the 2nd point above, I've briefly looked at outputs of TensorRT PNet, RNet and ONet.  I think I've observe the problem in ONet outputs.  More specifically, output of the first image crop in the batch looks correct and the others in the same batch look problematic...

Regarding the 3rd point above, I've seen similar issues reported by quite a few other people:

* [TensorRT partially corrupt output for batch size >1](https://forums.developer.nvidia.com/t/tensorrt-partially-corrupt-output-for-batch-size-1/168511)
* [TensorRT 7.1 produce incorrect results when batch size >1](https://github.com/NVIDIA/TensorRT/issues/842)
* [C++ TensorRT Batch Inference gives wrong result](https://github.com/dusty-nv/jetson-inference/issues/280)
* [TensorRT context.enqueue gives wrong result for all frames other than first](https://stackoverflow.com/questions/52856362/tensorrt-context-enqueue-gives-wrong-result-for-all-frames-other-than-first)

So this seems to be an issue of the latest TensorRT library itself.  If I later find a solution/workaround, I would update the code and post here.

jkjung-avt, 2021-03-01:
Here is an update of this particular issue.

**Problem description:**  In a sequence of image frames, the TensorRT MTCNN detects all faces in the first frame correctly but only detects (at most) 1 face in subsequent frames.  I could reproduce this issue with TensorRT 7.1.3.4 (e.g. JetPack-4.4).

**Additional observations:**  As stated in https://github.com/jkjung-avt/tensorrt_demos/issues/313#issuecomment-780276373, the problem does not occur on TensorRT 3.x and 4.x.

**2021-03-01 update:**  I tested the same TensorRT MTCNN code with TensorRT 7.2.2.3 on my x86_64 PC, and did not see the problem.

Based on these test results, I'd conclude that this is **a bug in TensorRT 7.1** and has been **fixed in TensorRT 7.2**.  As [reported by many other people](https://github.com/jkjung-avt/tensorrt_demos/issues/313#issuecomment-780276373), TensorRT 7.1 seems to produce wrong inference results for batch>1 inputs in some circumstances. 
 For people encountering this problem, please try to use another version of TensorRT than 7.1.

I'll close this issue if there's no further comment in a while.

closed 2021-03-08 by jkjung-avt