jkjung-avt/tensorrt_demos #314: Docker support for this repo

rahulsingh701, 2020-12-31:
Hi do you have a docker file for running this demo.If so can you please share.
If you do not have a docker file for this them can you please tell me how can I install Bazel In docker file getting stuck at that part

jkjung-avt, 2020-12-31:
I don't have a docker file for this repo.

Why do you need to install bazel inside docker?  What error do you encounter?

rahulsingh701, 2020-12-31:
Because when I try to run your install tensorflow 1.12.2 sh script it’s
shows an error bazel not installed inside docker

jkjung-avt, 2020-12-31:
"install_tensorflow-1.12.2.sh" is in my [jkjung-avt/jetson_nano](https://github.com/jkjung-avt/jetson_nano) repository, not this one.  I don't think you need to run it for this repo...

Try to use a Docker image with TensorRT provided by NVIDIA ([NGC](https://ngc.nvidia.com/catalog/containers?orderBy=modifiedDESC&pageNumber=0&query=&quickFilter=containers&filters=)) instead.

rahulsingh701, 2020-12-31:
So to run the ssd demo I don’t need tensorflow only tensor rt ?

jkjung-avt, 2020-12-31:
If you'd like to build TensorRT engines from the TensorFlow SSD models, you will need to install tensorflow.  You should be able to do that in the docker container by simply `pip install tensorflow==<verison>`.

closed 2021-01-06 by jkjung-avt