jkjung-avt/tensorrt_demos #316: mtcnn example fail on running

hubalaga, 2021-01-03:
Hi

When i try to run this example i get this error....

Parameter check failed at: runtime.cpp::deserializeCudaEngine::30, condition: (blob) != nullptr
Assertion fail in file 'trtNet.cpp' line 146: _engine is null
terminate called after throwing an instance of 'std::exception'
  what():  std::exception
Aborted

jkjung-avt, 2021-01-04:
> When I compile mtcnn there is some issues:
> ......

Those warning could be ignored.

Did you execute `./create_engines` successfully?  It should produce the TensorRT engine files, "det1.engine", "det2.engine" and "det3.engine" in the "mtcnn/" directory.

hubalaga, 2021-01-05:
You're absolutely right, there was a problem with that. It works beautifully now and I am absolutely impressed with the speed (~ 14FPS!). Thanks a lot!!

closed 2021-01-05 by hubalaga