jkjung-avt/tensorrt_demos #317: Support for TensorRt 5

tushardhadiwal, 2021-01-04:
@jkjung-avt Hello, Thank you for your great work.
Is it possible to adds support for TensorRT 5 for Jetson nano?
It is not possible for me to upgrade to tensorrt 6+ as there are other modules runnning which need tensorrt 5.

Thanks

jkjung-avt, 2021-01-04:
If you are using TensorRT 5.1, then the required modification should not be much.  I think you would just need to modify the `YoloLayerPlugin` class as a subclass of `IPluginV2Ext` (instead of `IPluginV2IOExt`) and simplify/remove unnecessary methods.

https://github.com/jkjung-avt/tensorrt_demos/blob/c26b81a5d2d12e9e407ba34cd5a110983c60ec65/plugins/yolo_layer.h#L38

I don't have an older JetPack environment for development/testing so I'm reluctant to work on that...

tushardhadiwal, 2021-01-04:
> If you are using TensorRT 5.1, then the required modification should not be much. I think you would just need to modify the `YoloLayerPlugin` class as a subclass of `IPluginV2Ext` (instead of `IPluginV2IOExt`) and simplify/remove unnecessary methods.

@jkjung-avt I am running TensorRt 5.0.6.3, with CuDNN 7.3.1.28 and CUDA 10.0.166

Will modifying above class still work?

jkjung-avt, 2021-01-04:
Please refer to [TensorRT 5.1.5 Release Notes](https://docs.nvidia.com/deeplearning/tensorrt/release-notes/tensorrt-5.html):

> New supported operations
> ......
> ONNX: Added ArgMax, ArgMin, Clip, Cast, Elu, Selu, HardSigmoid, Softplus, Gather, ImageScaler, LeakyReLU, ParametricSoftplus, Sin, Cos, Tan, Asin, Acos, Atan, Sinh, Cosh, Asinh, Acosh, Atanh, Ceil, Floor, ScaledTanh, Softsign, Slice, ThresholdedRelu and Unsqueeze ops.

ONNX ops such as "Softplus", "LeakyReLU" and "Slice" are required for yolov4/yolov4-tiny models.  They are not available in TensorRT 5.0.x.  So you won't be able to build the TensorRT yolov4/yolov4-tiny engines directly with the ONNX parser in TensorRT 5.0.x.  (You'll have to modify the parser by yourself or even create plugins for unsupported ops, which could be a lot of work...)

tushardhadiwal, 2021-01-04:
@jkjung-avt I have already created the .trt file of the model using my jetpack 4.4.1 environment. So If I plan to directly use this .trt file created with TensorRT 7.1.3, on board with TensorRt 5.0.x, Do I have to worry about Onnx parser?

jkjung-avt, 2021-01-04:
> I have already created the .trt file of the model using my jetpack 4.4.1 environment. So If I plan to directly use this .trt file created with TensorRT 7.1.3, on board with TensorRt 5.0.x

That would not work.  You need to build the TensorRT engine using the same version you intend to do inference with.

closed 2021-01-08 by jkjung-avt