jkjung-avt/tensorrt_demos #318: How to use own class in TensorRT YOLOv4

PingHan88, 2021-01-04:
I have a trained model, but how do I change the class name?

jkjung-avt, 2021-01-04:
Modify the `get_cls_dict()` function for your own custom model.

https://github.com/jkjung-avt/tensorrt_demos/blob/8b81bbc3921da74b59731ab77a3fb2d33ceb0f13/trt_yolo.py#L94

https://github.com/jkjung-avt/tensorrt_demos/blob/8b81bbc3921da74b59731ab77a3fb2d33ceb0f13/utils/yolo_classes.py#L99-L104

PingHan88, 2021-01-04:
Thanks for your reply,very thank you

closed 2021-01-05 by jkjung-avt

PiyalGeorge, 2021-01-15:
Hi @jkjung-avt , I have a doubt and i wish to communicate with you for a good suggestion. **My doubt has no relation with the about issue** . I need to run a Gender detection in **nano** or **xavier-nx** board with good accuracy and FPS. I've searched but i couldn't find one. Can you kindly suggest me any good methods/repos for this purpose?

jkjung-avt, 2021-01-15:
@PiyalGeorge  I don't have any experience using human gender classification models.  So I won't be able to provide any suggestion on that.