jkjung-avt/tensorrt_demos #320: Errors on running the scripts after few successful attempts (openCV errors after terminating MJPG scripts)

dvitrano, 2021-01-04:
Hi, I cannot exactly link the behaviour in the subject to this root cause but according to the facts - at least on my side - it seems after running successfully a couple of times the "mjpg" version of the trt_yolo scripts now the system is not able to  execute any of the demos due to some random issues at openCV/GStreamer level, all seeming to converge into "open capture device" issue.
As far as it count in Linux, I also restarted the system but no luck.

Error messages are basically the same. Here is the output when trying 
$ python3 trt_yolo.py --onboard 0 -m yolov3-416

Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:568 Failed to create CaptureSession
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
WARNING:root:Camera: cap.read() returns no image!
ERROR: failed to open camera!

a more verbose output when trying with a RTSP source
$python3 trt_yolo.py --rtsp rtsp://admin:pwd@192.168.20.12:554/live/0/MAIN -m yolov3-416
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module udpsrc0 reported: Internal data stream error.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
WARNING:root:Camera: starting while cap is not opened!
ERROR: failed to open camera!

I don't have a USB webcam for more tests.

I'm also happy to provide more logs or outputs if needed.

Last resort I would need to rebuild everything from scratch ?
Any hint would be appreciated.
Thanks !

dvitrano, 2021-01-04:
I checked openCV version

Python 3.6.9 (default, Oct  8 2020, 12:12:24)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2.__version__
'4.1.1'
>>>

also:
pkg-config --modversion opencv4
4.1.1

dvitrano, 2021-01-04:
more stuff..

Python 3.6.9 (default, Oct  8 2020, 12:12:24)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cap.isOpened()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'cap' is not defined
>>> cap = cv2.VideoCapture(0)
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created

jkjung-avt, 2021-01-05:
Please check first whether your cameras work with the "gst-launch-1.0" commands.

```
$ gst-launch-1.0 nvarguscamerasrc ! \
                 ‘video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)NV12, framerate=(fraction)30/1' ! \ 
                 nvegltransform ! nveglglessink -e
$ gst-launch-1.0 rtspsrc location=rtsp://admin:pwd@192.168.20.12:554/live/0/MAIN ! \
                 rtph264depay ! h264parse ! omxh264dec ! \
                 nveglglessink -e
```

Reference: [Camera Capture with GStreamer-1.0](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Faccelerated_gstreamer.html%23wwpID0E0ZQ0HA)

dvitrano, 2021-01-08:
JK, first of all thanks for your time !
first command worked ok
OUTPUT:
dvitrano@domenico-jetson:~/jetson_nano/tensorrt_demos$ gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)NV12, framerate=(fraction)30/1' ! nvegltransform ! nveglglessink -e
Setting pipeline to PAUSED ...

Using winsys: x11 
Pipeline is live and does not need PREROLL ...
Got context from element 'eglglessink0': gst.egl.EGLDisplay=context, display=(GstEGLDisplay)NULL;
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21,000000 fps Duration = 47619048 ; Analog Gain range min 1,000000, max 10,625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28,000001 fps Duration = 35714284 ; Analog Gain range min 1,000000, max 10,625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29,999999 fps Duration = 33333334 ; Analog Gain range min 1,000000, max 10,625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29,999999 fps Duration = 33333334 ; Analog Gain range min 1,000000, max 10,625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59,999999 fps Duration = 16666667 ; Analog Gain range min 1,000000, max 10,625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120,000005 fps Duration = 8333333 ; Analog Gain range min 1,000000, max 10,625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 2 
   Output Stream W = 1920 H = 1080 
   seconds to Run    = 0 
   Frame Rate = 29,999999 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
ERROR: from element /GstPipeline:pipeline0/GstEglGlesSink:eglglessink0: Output window was closed
Additional debug info:
/dvs/git/dirty/git-master_linux/3rdparty/gst/gst-nveglglessink/ext/eglgles/gsteglglessink.c(882): gst_eglglessink_event_thread (): /GstPipeline:pipeline0/GstEglGlesSink:eglglessink0
EOS on shutdown enabled -- waiting for EOS after Error
Waiting for EOS...
Terminated

Then i tried RTSP source no luck:

dvitrano@domenico-jetson:~/jetson_nano/tensorrt_demos$ gst-launch-1.0 rtspsrc location=rtsp://admin:XXX@192.168.20.12:554/live/0/MAIN ! rtph264depay ! h264parse ! omxh264dec ! nveglglessink -e
Setting pipeline to PAUSED ...

Using winsys: x11 
Pipeline is live and does not need PREROLL ...
Got context from element 'eglglessink0': gst.egl.EGLDisplay=context, display=(GstEGLDisplay)NULL;
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://admin:admin@192.168.20.12:554/live/0/MAIN
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (request) SETUP stream 1
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Delayed linking failed.
Additional debug info:
./grammar.y(510): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0:
streaming stopped, reason not-linked (-1)
[... 45 lines cut ...]
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
WARNING:root:Camera: cap.read() returns no image!
ERROR: failed to open camera!

and also GST-LAUNCH:

dvitrano@domenico-jetson:~/jetson_nano/tensorrt_demos$ gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)NV12, framerate=(fraction)30/1' ! nvegltransform ! nveglglessink -e
Setting pipeline to PAUSED ...

Using winsys: x11 
Pipeline is live and does not need PREROLL ...
Got context from element 'eglglessink0': gst.egl.EGLDisplay=context, display=(GstEGLDisplay)NULL;
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:568 Failed to create CaptureSession
Got EOS from element "pipeline0".
Execution ended after 0:00:00.222312113
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

I can't honestly say how I was able to go back into normal state since I opened the issue. I just did not use the board for few days.. I don't think Gstreamer has "capacitors" inside :) but looks like after a while Gstreamer was happy to load initially.
Now nothing works. I will try with some reboots or wait again some time.
it seems the abrupt termination of the command might leave  some  portion of the pipeline in  some broken state?

jkjung-avt, 2021-01-09:
> Then i tried RTSP source no luck:

Make sure your RTSP source (rtsp://admin:admin@192.168.20.12:554/live/0/MAIN) is working first.

> NOW trtyolo with onboard camera does not work anymore:
> ......
> and also GST-LAUNCH:

It's hard to draw any conclusion from this.  It might be a hardware related problem as well.

jkjung-avt, 2021-01-16:
Closing, since this doesn't seem to point to a bug in the code.  I'm not able to reproduce the problem, either.

closed 2021-01-16 by jkjung-avt