jkjung-avt/tensorrt_demos #321: Improving inference time

tushardhadiwal, 2021-01-07:
Hi,

I came across this trick for improving inference time. https://github.com/opencv/opencv/pull/14827#issuecomment-568156546
While converting yolov4.cfg to tensort engine file, The cfg file that Used did not have nms-threshold=0 set in all yolo layers.
I do see some code in this repo for nms boxes etc.

Will i get any speedup in inference time of yolov4 if I add the values in cfg file? Or is this already taken care of while building trt engine?

Thanks

jkjung-avt, 2021-01-07:
Thanks for raising the question.  But that does not apply to the code in this repo.

If you can read CUDA code, you'd see that I don't do NMS in the yolo_layer plugin.

https://github.com/jkjung-avt/tensorrt_demos/blob/8b81bbc3921da74b59731ab77a3fb2d33ceb0f13/plugins/yolo_layer.cu#L198-L249

Instead, I do NMS with python as shown below.  The NMS code is written in python and indeed could be slow.  You might improve FPS by optimizing this part (for example, replace it with C++ code).

https://github.com/jkjung-avt/tensorrt_demos/blob/8b81bbc3921da74b59731ab77a3fb2d33ceb0f13/utils/yolo_with_plugins.py#L139-L146

BigJoon, 2021-01-10:
Thanks for the detailed explanation. I'll think about optimizing the NMS code part with C++ code.
I think it will be a fun and proud work.

closed 2021-01-13 by jkjung-avt