jkjung-avt/tensorrt_demos #325: yolov4-csp-512.onnx to yolov4-csp-512.trt error

c7934597, 2021-01-13:
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: 173_convolutional_mish for ONNX node: 173_convolutional_mish
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: 173_convolutional_mish for ONNX tensor: 173_convolutional_mish
[TensorRT] VERBOSE: ModelImporter.cpp:179: 173_convolutional_mish [Mul] outputs: [173_convolutional_mish -> (1, 1024, 16, 16)], 
[TensorRT] VERBOSE: ModelImporter.cpp:103: Parsing node: 174_convolutional [Conv]
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 173_convolutional_mish
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 174_convolutional_conv_weights
[TensorRT] VERBOSE: ModelImporter.cpp:119: Searching for input: 174_convolutional_conv_bias
[TensorRT] VERBOSE: ModelImporter.cpp:125: 174_convolutional [Conv] inputs: [173_convolutional_mish -> (1, 1024, 16, 16)], [174_convolutional_conv_weights -> (18, 1024, 1, 1)], [174_convolutional_conv_bias -> (18)], 
[TensorRT] VERBOSE: builtin_op_importers.cpp:450: Convolution input dimensions: (1, 1024, 16, 16)
[TensorRT] VERBOSE: ImporterContext.hpp:141: Registering layer: 174_convolutional for ONNX node: 174_convolutional
[TensorRT] VERBOSE: builtin_op_importers.cpp:533: Using kernel: (1, 1), strides: (1, 1), prepadding: (0, 0), postpadding: (0, 0), dilations: (1, 1), numOutputs: 18
[TensorRT] VERBOSE: builtin_op_importers.cpp:534: Convolution output dimensions: (1, 18, 16, 16)
[TensorRT] VERBOSE: ImporterContext.hpp:116: Registering tensor: 174_convolutional for ONNX tensor: 174_convolutional
[TensorRT] VERBOSE: ModelImporter.cpp:179: 174_convolutional [Conv] outputs: [174_convolutional -> (1, 18, 16, 16)], 
程式記憶體區段錯誤

-----------------------------------------------------------------------------------------------------------------------------------------------------------
問題如上，使用電腦跟Xavier Nx jetpack 4.4來轉都失敗，請問onnx_to_tensorrt.py程式，貌似裡面沒支援yolov4-csp-512嗎?

jkjung-avt, 2021-01-14:
https://github.com/jkjung-avt/tensorrt_demos/blob/899770162fecc21475db38d670b5b3467874046e/yolo/download_yolo.sh#L25-L27

I used these cfg and pre-trained weights files, and was able to convert it to a TensorRT engine on Jetson Xavier NX without problem.  You could verify it.

c7934597, 2021-01-14:
Thank you for that, I use the cfg of darknet, and then I already could convert it to a TensorRT engine.

closed 2021-01-14 by c7934597