jkjung-avt/tensorrt_demos #326: RTSP link not working with authentication

chandu1263, 2021-01-14:
How to setup gstreamer with rtsp link with authentication.
I print the gst_elements
`avdec_dds: libav DirectDraw Surface image decoder decoder\nlibav:  avdec_dfa: libav Chronomaster DFA decoder\nlibav:  avdec_dirac: libav BBC Dirac VC-2 decoder\nlibav:  avdec_dnxhd: libav VC3/DNxHD decoder\nlibav:  avdec_dpx: libav DPX (Digital Picture Exchange) image decoder\nlibav:  avdec_dsicinvideo: libav Delphine Software International CIN video decoder\nlibav:  avdec_dvvideo: libav DV (Digital Video) decoder\nlibav:  avdec_dxa: libav Feeble Files/ScummVM DXA decoder\nlibav:  avdec_dxtory: libav Dxtory decoder\nlibav:  avdec_dxv: libav Resolume DXV decoder\nlibav:  avdec_eacmv: libav Electronic Arts CMV video decoder\nlibav:  avdec_eamad: libav Electronic Arts Madcow Video decoder\nlibav:  avdec_eatgq: libav Electronic Arts TGQ video decoder\nlibav:  avdec_eatgv: libav Electronic Arts TGV video decoder\nlibav:  avdec_eatqi: libav Electronic Arts TQI Video decoder\nlibav:  avdec_8bps: libav QuickTime 8BPS video decoder\nlibav:  avdec_escape124: libav Escape 124 decoder\nlibav:  avdec_escape130: libav Escape 130 decoder\nlibav:  avdec_exr: libav OpenEXR image decoder\nlibav:  avdec_ffv1: libav FFmpeg video codec #1 decoder\nlibav:  avdec_ffvhuff: libav Huffyuv FFmpeg variant decoder\nlibav:  avdec_fic: libav Mirillis FIC decoder\nlibav:  avdec_fits: libav Flexible Image Transport System decoder\nlibav:  avdec_flashsv: libav Flash Screen Video v1 decoder\nlibav:  avdec_flashsv2: libav Flash Screen Video v2 decoder\nlibav:  avdec_flic: libav Autodesk Animator Flic video decoder\nlibav:  avdec_flv: libav FLV / Sorenson Spark / Sorenson H.263 (Flash Video) decoder\nlibav:  avdec_fmvc: libav FM Screen Capture Codec decoder\nlibav:  avdec_4xm: libav 4X Movie decoder\nlibav:  avdec_fraps: libav Fraps decoder\nlibav:  avdec_frwu: libav Forward Uncompressed decoder\nlibav:  avdec_g2m: libav Go2Meeting decoder\nlibav:  avdec_gdv: libav Gremlin Digital Video decoder\nlibav:  avdec_gif: libav GIF (Graphics Interchange Format) decoder\nlibav:  avdec_h261: libav H.261 decoder\nlibav:  avdec_h263: libav H.263 / H.263-1996, H.263+ / H.263-1998 / H.263 version 2 decoder\nlibav:  avdec_h263i: libav Intel H.263 decoder\nlibav:  avdec_h263p: libav H.263 / H.263-1996, H.263+ / H.263-1998 / H.263 version 2 decoder\nlibav:  avdec_h264: libav H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 decoder\nlibav:  avdec_hap: libav Vidvox Hap decoder\nlibav:  avdec_h265: libav HEVC (High Efficiency Video Coding) decoder\nlibav:  avdec_hnm4video: libav HNM 4 video decoder\nlibav:  avdec_hq_hqa: libav Canopus HQ/HQA decoder\nlibav:  avdec_hqx: libav Canopus HQX decoder\nlibav:  avdec_huffyuv: libav Huffyuv / HuffYUV decoder\nlibav:  avdec_hymt: libav HuffYUV MT decoder\nlibav:  avdec_idcinvideo: libav id Quake II CIN video decoder\nlibav:  avdec_iff: libav IFF ACBM/ANIM/DEEP/ILBM/PBM/RGB8/RGBN decoder\nlibav:  avdec_imm4: libav Infinity IMM4 decoder\nlibav:  avdec_indeo2: libav Intel Indeo 2 decoder\nlibav:  avdec_indeo3: libav Intel Indeo 3 decoder\nlibav:  avdec_indeo4: libav Intel Indeo Video Interactive 4 decoder\nlibav:  avdec_indeo5: libav Intel Indeo Video Interactive 5 decoder\nlibav:  avdec_interplayvideo: libav Interplay MVE video decoder\nlibav:  avdec_jpeg2000: libav JPEG 2000 decoder\nlibav:  avdec_jpegls: libav JPEG-LS decoder\nlibav:  avdec_jv: libav Bitmap Brothers JV video decoder\nlibav:  avdec_kgv1: libav Kega Game Video decoder\nlibav:  avdec_kmvc: libav Karl Morton\'s video codec decoder\nlibav:  avdec_lagarith: libav Lagarith lossless decoder\nlibav:  avdec_loco: libav LOCO decoder\nlibav:  avdec_lscr: libav LEAD Screen Capture decoder\nlibav:  avdec_m101: libav Matrox Uncompressed SD decoder\nlibav:  avdec_magicyuv: libav MagicYUV video decoder\nlibav:  avdec_mdec: libav Sony PlayStation MDEC (Motion DECoder) decoder\nlibav:  avdec_mimic: libav Mimic decoder\nlibav:  avdec_mjpeg: libav MJPEG (Motion JPEG) decoder\nlibav:  avdec_mjpegb: libav Apple MJPEG-B decoder\nlibav:  avdec_mmvideo: libav American Laser Games MM Video decoder\nlibav:  avdec_motionpixels: libav Motion Pixels video decoder\nlibav:  avdec_mpeg2video: libav MPEG-2 video decoder\nlibav:  avdec_mpeg4: libav MPEG-4 part 2 decoder\nlibav:`
This is a part of that gst_elements.
`gst_str = (f'''
            rtspsrc location={uri} latency={latency} user-id={userid} user-pw={password} protocol=tcp+udp-mcast+udp !
            rtph264depay ! h264parse ! decodebin !
            nvvideoconvert !
            video/x-raw,format={frame_format} !
            fakesink name=s
        ''')`
I tried this by doing some research over stackoverflow, but still couldn't make it work.
Can you please share the snippet for how to make rtsp streams run with authentication?
TIY

jkjung-avt, 2021-01-14:
Couldn't you just embedded username and password in the RTSP URL?  For example, "--rtsp rtsp://admin:123456@192.168.1.1/live.sdp".

chandu1263, 2021-01-14:
I tried the same, but I don't know why it is not working. I thought the problem is with authentication.
I am not sure why it is not working.

jkjung-avt, 2021-01-14:
Please make sure your RTSP source is working with other tools (e.g. VLC) first.

chandu1263, 2021-01-15:
Yup it is working properly. If I try using cv2.VideoCapture(link)
Is deep stream necessary for running this code on a tensorrt docker?
Is there any specific installation steps to make this run (opencv, deepstream, gstreamer)?

jkjung-avt, 2021-01-15:
Please refer to [How to Capture and Display Camera Video with Python on Jetson TX2](https://jkjung-avt.github.io/tx2-camera-with-python/).  You should make sure your opencv supports gstreamer pipelines.  In addition, you might need to install some additional gstreamer plugins.

And my code does not depend on DeepStream at all.

Otherwise, if you still think the problem is related to RTSP user authentication, I probably won't be able to help further.

jkjung-avt, 2021-01-24:
Closing since this does not seem to point to a bug in the code of this repo.

closed 2021-01-24 by jkjung-avt