jkjung-avt/tensorrt_demos #327: Failed to build TensorRT engine because the weights are outside f16 range.

jrnladhad, 2021-01-15:
Thank you for such a useful tool. I have a custom YOLOv3 model that I trained using weights from the official website and then fine-tuned them to adjust to my data and network architecture using PyTorch. Once I obtained the new weights I was able to generate the onnx format using the script provided by you but fail to create the tensorRT engine.

jkjung-avt, 2021-01-15:
https://github.com/jkjung-avt/tensorrt_demos/issues/161#issuecomment-662473343

closed 2021-01-18 by jkjung-avt

jrnladhad, 2021-01-20:
Hey, 
I retrained the model and lowered the decay value during the training, earlier it was 0.0005 and this time it was 0.005 and yet I am running into the same issue.

jkjung-avt, 2021-01-21:
You could try to lower learning rate (while keeping the same weight decay value) and train the model for a longer period of time.

If that still doesn't work, then you should probably consider using FP32 TensorRT engine.