jkjung-avt/tensorrt_demos #328: custom yolov4-tiny model runs slower than expected on jetson nano

IbrahimBond, 2021-01-15:
i have converted a yolov4-tiny model to TRT successfully, but i am getting 10 fps only, which is much lower than the expected result.

i have realized that the TRT model size is 32.3 mbs, when the original model is 23.5mb in size. doesnt converting to TRT reduce the model size? 

also do i need to do something prior to converting the model to make it work in 16fp or does the conversion process do that automatically?

thanks

PiyalGeorge, 2021-01-15:
Reducing network size (width and height) can help to a limit. But doing that can reduce the accuracy level of distant objects.

jkjung-avt, 2021-01-15:
@IbrahimBond  Are you referring to a custom yolov4-tiny model?  How is it different from the pre-trained darknet yolov4-tiny (COCO) model?

jkjung-avt, 2021-01-16:
> @jkjung-avt yes i am referring to a custom model i trained. only difference is that it is trained to detect one class. so i have 18 filters in the layer before yolo layer with changing number of classes.

I converted the pre-trained yolov4-tiny-416 model to a TensorRT engine on my Jetson Nano DevKit.  I could get 25.5 FPS when testing it.  And the trt file size (13MB) is roughly half of the original weights file (24MB).

closed 2021-01-24 by jkjung-avt