jkjung-avt/tensorrt_demos #330: Training yolov4-tiny with higher resolution, doing inferencing with a lower resolution

aeozyalcin, 2021-01-19:
Hello,

I trained a custom yolov4-tiny object detection model using Darknet with 512x512 resolution, and I noticed that if I lower the inferencing resolution slightly within Darknet to 480x480, the map increases (less false positives). Sounds like this is a perfectly acceptable thing to do with yolo models, as long as training & inferencing resolution is within ~40% ([source](https://github.com/AlexeyAB/Yolo_mark/issues/25#issuecomment-363252861))

I am trying to replicate the same thing within TensorRT using your library, where I lower the resolution to a value lower than what I trained the model on. To do this, I tried:

- Keep width/height at 512x512 in the cfg file, but change the model name so it has 480x480 ==> This gives an error during onnx to trt conversion.
- Change width/height to 480x480 in the cfg file, and change the model name so it has 480x480 ==> conversion is successful, however the model does not detect any objects at all 
- Keep 512x512 in cfg file, and in the model name, then do yolo to onnx, and onnx to trt conversions, then change the name of the resulting .trt file to have 480x480 ==> model does not detect any objects at all 

Do you have any insights as to how I can achieve changing the network resolution? 

Thanks!

jkjung-avt, 2021-01-19:
> Change width/height to 480x480 in the cfg file, and change the model name so it has 480x480 ==> conversion is successful, however the model does not detect any objects at all

This is how it supposed to be done.

My code in this repo actually demonstrates it.  In the step-by-step guide, we download the pretrained darknet yolov4-608x608 model and generates 608x608, 416x416 and 288x288 TensorRT engines using the same weights.  All these TensorRT engines seem to detect objects correctly.

Let me know how I could reproduce the issue and I will take a look.

aeozyalcin, 2021-01-19:
Hi JK,

Thanks for the quick response! I tried that method again, and it did work this time. I must have done something wrong the first time I tried it.

Thanks!

closed 2021-01-19 by aeozyalcin