jkjung-avt/tensorrt_demos #331: Makefile has directories which I don't have.

MatsalVistry, 2021-01-19:
The makefile in the plugins directory has TENSORRT_INCS=-I"/usr/local/TensorRT-7.1.3.4/include" while I don't even have /usr/local/TensorRT*. Is there something I'm missing or is there anything I have to change?

jkjung-avt, 2021-01-20:
> TENSORRT_INCS=-I"/usr/local/TensorRT-7.1.3.4/include"

This is where I (manually) installed TensorRT libraries on my x86_64 PC.  It does not hurt if the path does not exist.

In case you have TensorRT libraries installed at another path, be sure to modify the Makefile accordingly.

MatsalVistry, 2021-01-20:
So I ran 
$ sudo dpkg -i nv-tensorrt-repo-${os}-${tag}_1-1_amd64.deb
$ sudo apt-key add /var/nv-tensorrt-repo-${tag}/7fa2af80.pub
$ sudo apt-get update
$ sudo apt-get install tensorrt
$ sudo apt-get install python3-libnvinfer-dev

and I can't find any libs or include folder. Do I just leave it as it is and it should work?

jkjung-avt, 2021-01-20:
If you are using a Jetson device, TensorRT should have been installed by JetPack already.

If you are using a x86_64 PC, refer to the official [TensorRT Installation Guide](https://docs.nvidia.com/deeplearning/tensorrt/install-guide/index.html).

When you install TensorRT with `apt-get install`, I think the library files would be located at "/usr/lib/x86_64-linux-gnu".  And you probably don't need to make any modification to the Makefile.

MatsalVistry, 2021-01-20:
Got it, thank you so much!

closed 2021-01-21 by jkjung-avt

LiDe2000, 2025-01-15:
> If you are using a Jetson device, TensorRT should have been installed by JetPack already.
>
> If you are using a x86_64 PC, refer to the official [TensorRT Installation [...]

I am using a Jetson Orin Nano, and TensorRT is already installed, but I cannot find the header file path.